opentx/opentx #6340: Without LiPo-battery radio writes files that are seen as 1-1-2098

notlebowski, 2019-02-24:
My Horus X12S first production run had an empty battery.
Because the GPS sets the time after it locks on some satellites, I didn't hurry to replace the battery.

But this changed after I wound up on the field for a maiden flight and my rudder was reverse.
It turned out my Horus had an old setup.
I thought it was strange because the battery is, afaik, not for things like setups.

Today I found the reason for it.
It turns out my SD-Card saved stuff as 1-1-2098 when it was running with a reset clock.
Because I'm using sync-software to make sure the SD-Card content is the same on several computers (with Syncthing) these files get replaced whenever I do a resync.

I'm not sure what the best solution is to prevent this.
I would totally understand if you don't want to solve this because one should of course not run the radio with a defective RTC-battery.
I know that I will not have this problem for another 3 years.
Just wanted to let you know.

[image]

mpaperno, 2019-02-24:
FWIW, the Companion SD sync function in branch 2.3 code will ignore (with error) any files with future dates on either side (if date compare was enabled). #6009 Guess it never made it into 2.2 branch because 2.3 was going to be out "soon" anyway.

3djc, 2019-02-25:
Unless something really unexpected pops up, 2.3 should be the next version we will release (and yes, asking for an ETA is pointless :))

closed 2019-02-25 by notlebowski

3djc, 2019-02-25:
Not you specificaly, but each time we talk about next version, we get bombarded with email asking when :)

notlebowski, 2019-02-25:
I never asked for Eta's nor do you post this in every ticket. Therefore it a clearly personal.
I have no need to get this fixed other than to improve opentx. The new battery in my Horus will not break down for another 3 years, so all I did was helping others out that may soon bump into this as well.

notlebowski, 2019-02-25:
Maybe I overreacted as well. 
But things would really improve if the devs didn't treat posters that post tickets as spoiled brats that only want more and immediately.
I know who I am and I'm someone that likes to help people out and also like to improve projects by posting issues I bump into. 
Most of the time I don't have anything to gain personally.

kilrah, 2019-02-25:
> Therefore it a clearly personal.

No it's just you being paraniod as usual given that it was just a general comment, no reason to take it as specifically directed to you.

notlebowski, 2019-02-25:
Yeah, right...

schwabe, 2019-02-25:
Read through other issues and you will see. That remark of not asking for ETA is more for other users who stumble upon this issue. I can undertsand that you felt that it was directed at you but if you now the rest of the tickets you also would not feel that way.

notlebowski, 2019-02-25:
I have no problem to admit I was overreacting, but that there's a certain hostility is not something I just make up. Several of my tickets are just ignored. 
Not even a comment....

https://github.com/opentx/opentx/issues/6293
https://github.com/opentx/opentx/issues/6235
https://github.com/opentx/opentx/issues/6198
https://github.com/opentx/opentx/issues/6196

kilrah, 2019-02-25:
Why do you want us to tell you to stop posting issues? You're a grown man, you can choose to do whatever you please on your own.

There are multiple possible reasons for us to either accept or reject suggestions, act or not on them, or even comment or not on them in a given timeframe. Maybe some will get relevant and commented on in a year when other developments are done.
If you think you're not getting the attention or feedback you're looking for, then by all means take the actions you deem appropriate. If that means not posting issues anymore, no problem... but that's on you, not us.

The only thing we'd appreciate you to "stop" is wasting our time with your "special snowflake" attention-seeking drama that never has anything to do with the issue itself.

The actual issue here was resolved in the first answer, 2.3 will take care of it. The rest is just a waste of time because you somehow feel offended by a general comment about our release schedules for no reason...