opentx/opentx #6341: Flex & Eu

scott-adamson1975, 2019-02-25:
Possibility of adding option to include R9M/R9MLite flex support **AND** EU for ijxt?

each option cancels out the other so it is not possible to choose both.

i realise that there are LGBT/Euro etc. implications or whatever they are called but a responsible user would limit his power output much like openLRS....

3djc, 2019-02-25:
Given EU option only hides D8 choice from the UI (and does nothing else), if you fly Flex, you might as well show the D8 option.

The reason why they are excluding themselves is that EU option is for the ones that what to make sure they cannot disobey the law, which doesn't make sense if you fly flex !

closed 2019-02-25 by 3djc

scott-adamson1975, 2019-02-25:
ok, reason behind it was that i can be legal when i'm at my flying club flying 3rd person, but when i go rogue and go for a little bit of fpv in the wilderness i can switch out to the extra range r9m module.

can using non eu with the ijxt running the eu firmware cause problems?

3djc, 2019-02-25:
As I said, eu option only hides d8 option from the ui, it does not touch in any way the way your xjt works. You can run fcc on your xjt while having eu selected, you can fly lbt without eu option. Once again “it only hides d8 choice”, nothing more