opentx/opentx #6346: how do I get more than 3 flightmodes?

jensdk, 2019-03-01:
The issue tracker here is for devlopment of OpenTX. This is not a
support site, only a bug tracking system. Please use support groups
like [rcgroups](https://rcgroups.com/) or
[openrcforum](https://www.openrcforums.com/forum/). See also our
[manual](https://opentx.gitbooks.io/manual-for-opentx-2-2/).

For quick questions you can also join our general chat.

Issues that fall into the category of asking for help, will be
generally closed as invalid without an answer.

jensdk, 2019-03-01:
In OTX above 2.2 it seems that it's only possible using 3 flightmodes the "easy" ...  on a 3pos switch, when I want to activate the  4.  mode, nothing happens.. 
i fly f3K planes  need 3 modes for camber and 1 for elevatorpreset when launching..  I have the 3 modes on SA and want the 4 on SF ..
using the "old" method doesn't seem to work.

3djc, 2019-03-01:
I'm sorry but this is a bug tracker, not a user support site. Please go to community places like rcgroups or openrcforums for usage type questions.

closed 2019-03-01 by 3djc

jensdk, 2019-03-01:
IT's a kind of bug if you read it correctly.. 
Flightmodes has been limited, since earlier version.... ?

3djc, 2019-03-01:
No, you have exactly the same number of flighmodes

jensdk, 2019-03-01:
Did you test it ?? 
I can get 3 modes, but when I want to activate the next "overule"  nothing happens.... 
Has the setup of modes been  changed since OTX 2.1 ?

kilrah, 2019-03-01:
No. You're just misunderstanding the priority order of modes.

mpjf01, 2019-03-02:
He seems to be claiming that something that used to work with a previous firmware now does not.  It might help him to understand why if he could be advised what was changed in the firmware that might have caused this issue for him.  It would help to see his .otx file for the model he is concerned about.

kilrah, 2019-03-02:
He got answers to his question on forums as expected.