opentx/opentx #6347: Load global volume functions at boot

nmaggioni, 2019-03-01:
The volume on my Horus X10S is adjusted with S2 via a Global Function, but it only kicks in after both the welcome sound and the eventual alerts have been played.

Having volume-related GFs being loaded before the splash screen could help in avoding "loud" sounds at boot: am I the only one tinkering with its radio late in the evening?

_This might be a subset of #2777, which seems abandoned._

3djc, 2019-03-01:
You can set the volume at boot in your radio setup screen, which would be used for before your volume function kicks in.

Functions are only run after safety checks have happened for obvious security reason

nmaggioni, 2019-03-01:
@3djc Thanks, I didn't think of that.

What about loading only functions that act on the volume? This way even startup sounds could be dynamically adjusted.

3djc, 2019-03-01:
For any functions to run, including volume, a lot of other things are needed, like mixer, which cannot be run before checks.

closed 2019-03-01 by nmaggioni