opentx/opentx #6348: File system in "Special Functions" not working

exocode, 2019-03-01:
**Describe the bug**
The file listing is not working in special functions. In Special Functions, when I try to select a file while "Play Track" is selected, I don't get a list of files.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Special functions'
2. create  a new entry
3. select "Play track"
4. try to select a file
5. See error (attached picture)

(BTW: I can select the files correctly from within OpenTX Companion.)

**Expected behavior**
In step 4 of "To Reproduce" a list of files should appear
**Screenshots**
[image]
[image]
[image]

**General information**
 - OpenTX version (2.2.3 aee5ac05b)
 - Hardware (Frsky Taranis Q X7)
 - OpenTX Companion (Version 2.2.3, Jan 7 2019 on MacOS)

kilrah, 2019-03-01:
Do you actually have the proper files at the right place, and the correct sound language selected?

exocode, 2019-03-01:
Hi, 
I am working in "english" (en) folder.
Simply dropped the OpenTX card content on it. added Amber sound pack, that's it..

exocode, 2019-03-01:
Could it be, that I am using Disk Content Version 2.2.2 and using OpenTX 2.2.3? (I didn't find any 2.2.3 Disk content)

3djc, 2019-03-01:
I cannot reproduce, works fine here. Have you checked there is not file (possibly hidden) with forbiden/strange name ?

exocode, 2019-03-01:
Thank you for your time!

I got it working:
- Erased EVERYTHING on the disk
- downloaded 2.2.3 Files again https://downloads.open-tx.org/2.2/release/sdcard/opentx-x7/
- uploaded it: Worked

Sorry for waisting time :-|

closed 2019-03-01 by exocode