opentx/opentx #6351: Developers formum?

guycnicholas, 2019-03-02:
I have finally worked through getting setup to build on my Windows box. I did have a few troubles along the way, for example, I had a number of errors with `from PIL import Image`, and after google'ing that a bit I found that updating from `PIL` to `Pillow` solved my problem. 
At one point I was able to launch the app and open my existing OpenTx files, but now as soon as `newFile()` attempts to construct a `RadioData` I get an exception on ` PROPERTYSTR2(display,       "Display")`.
I would really like to "give back" to the project, and what I am wondering is, is there a forum or a Slack channel or something where the developers chat, or is everything done here?

jtaylor2, 2019-03-02:
I brought up that error on the chat channel back on July 29, 2018 and provided the error report and a git blame of which commit broke it but they told me it was to early in 2.3 development to report it so I let it go.  Don't know if they still consider it too early or not.
https://chat.open-tx.org/channel/OpenTX_General?msg=bCYAztd9SA7e2QLeD

I compiled on both Windows 10 and openSUSE 15.0 Linux  this morning and it's still there on both for me to.  I looked at it but that qt profile code is beyond me.

Jim

3djc, 2019-03-02:
2.3 is currently really a wip at this stage, which is why we do not even do nightlies for it. We are currently working on a lot of changes. When those get merged, things will probably be even worse for a brief period before they improve. When we start doing 2.3 nightlies, we will start fixing issues on that branch

closed 2019-08-19 by kilrah