opentx/opentx #6352: Save stick mode with individual model

Jetrell, 2019-03-02:
I would like to make a feature request.

I use two different Stick Modes when flying between Heli's/Multicopters and Fixedwings.
Mode 1 for Fixedwings and Mode 3 for Heli's/Multicopter.

Presently. Once a model is loaded, I then have to go to the radio setup and change the stick mode selection as well.
If I forget this step. Which I once did... It will lead to a crash.

This is why I would like to request that....What even stick mode is selected at the time a model is saved. That the stick mode also be saved with that individual model.

Jetrell, 2019-03-19:
Looks like I must be the only person that fly's with 2 different stick modes ?

MarkusPfaff, 2019-03-27:
I'd try to use the mode setting for the main mode you use (meaning the one which applies to more models) and use appropriate input lines to use source ail for input rud and rud for input ail. You also need to redirect the trim sources also. Be careful with trim settings, i.e. test them thorougly before actual flying, because OpenTX has some flaws with "Throttle Trim Idle Only" and "Instant Trim" in 2.2.3 when redirecting trim to other sources (which you will need in your case).

Jetrell, 2019-04-05:
@MarkusPfaff 
Thanks for the idea. But as you said...It isn't without the possibility of a something being over looked.

That's why I was hoping for the simplicity and ease of the Stick Mode that was selected at that time, also being backed up and saved with that specific model.

Jetrell, 2019-12-25:
I'm still happy for this feature to be implemented if anyone gets the time :)

MarkusPfaff, 2020-06-23:
Still the option would be nice to have.

Jetrell, 2020-12-21:
Its not stale for me. I still think it would be a great feature.
It also make sense to have the Stick mode saved in the individual model memory and not have to remember to select it later.

Jetrell, 2021-06-19:
Im still interested in this feature.  There are bound to be many people out here that use more thsn one stick mode with different models.

lshems, 2021-06-20:
Look, there is no sense for changing stick modes actually.

Opentx doesn't care at all what stick mode you use.

The only thing which is relevant is which stick you use for throttle. And only if you use things as throttle trim.

But since both your modes use the same stick for throttle, there is really no reason at all to switch modes.

You can create four inputs, and name them according to your preferences. Then just put the correct stick source to that named input.

Use the inputs on your mixers, and not the direct stick sources, and you're ready to go.

There is nobody interested in fixing this, as nothing is broken. Just use opentx as it is intended, with an inputs layer mapping stick sources, whatever name they may have, to the mixes (outputs) you want

mpjf01, 2021-06-20:
The fundamental issue here is that the OP does not understand how OpenTx uses the mode setting. 

It is only used when a new model is created. 

Once a model has been created, changing the transmitter's mode will have no effect on that, or any other, existing model. 

Thus there is no need to change the setting when changing to a model using a different mode, and there is no reason to fear a crash if you do not.

lshems, 2021-06-21:
If the poster doesn't  understand the way opentx is working, your remark doesn't help. Better explain him how to get what he wants then, which I tried.

If my explanation on how to achieve what he want, improve it.

Address the poster, not me.

Jetrell, 2021-06-21:
@Ishems @mpjf01 
Guys. I never said anything was broken. I just thought it would be a good idea to integrate the `stick mode` into the individual model setup, As other radio's transmitter software does.

I get where you're coming from. I understand how to configure OpenTX to allocate the specific stick input to the appropriate output.  
But this misses the point. There are 4 stick mode selections in the `Radio Setup menu`. 
They purpose is clear.... So two people that fly different stick modes can fly the same model with the same transmitter, just by switching it over in the menu.
If I set the IO mixer for my own stick mode use, this prevents my flight buddy's, who use different stick modes from using my TX and models.

lshems, 2021-06-21:
I still don't get your problem.

?

What do you want to solve?

lshems, 2021-06-21:
Rereading and rereading again, and if I'm right the issue has to do with EXCHANGING models between radios, and thus having a mix of modes on the setup of several models in one radio due to copies of the other (different mode) radio

lshems, 2021-06-21:
Forget it. it is not clear for me.

Jetrell, 2021-06-21:
**@Ishems**  I would like each model memory to have the `Stick Mode` that it has selected under `Radio Setup menu`. Saved with its model model memory. When `backup model` is entered.

This way. The **primary user** of that transmitter, will know that the `Stick mode` that he uses for that specific type of model, will be loaded when `Select Model` is entered.
Rather than having to load the model for the memory... Then go into the `Radio Setup menu` and change the Stick mode too.... Which can be over looked.

e.g.
**Model Selection**
* 01 multi-rotor  - The owner of the transmitter uses `Stick mode 2` for that model. _Save_ `Stick Mode 2` with `backup model`.
* 02 Plane           - The owner of the transmitter uses `Stick mode 1` for that model. _Save_ `Stick Mode 1` with `backup model`.

But if the owner of the transmitter wants to let someone else fly his model. And they use a different stick mode... It will obvously have to be changed in under the `Radio Setup menu`.

Jetrell, 2021-06-21:
**@Ishems**  I would like each model memory to have the `Stick Mode` that it has selected under `Radio Setup menu`. Saved with its model model memory. When `backup model` is entered.

This way. The **primary user** of that transmitter, will know that the `Stick mode` that he uses for that specific type of model, will be loaded when `Select Model` is entered.
Rather than having to load the model for the memory, then go into the `Radio Setup menu` and change the Stick mode too.... Which can be over looked.

e.g.
**Model Selection**
* 01 multi-rotor  - The owner of the transmitter uses `Stick mode 2` for that model. _Save_ `Stick Mode 2` with `backup model`.
* 02 Plane           - The owner of the transmitter uses `Stick mode 1` for that model. _Save_ `Stick Mode 1` with `backup model`.

But if the owner of the transmitter wants to let someone else fly his model. And they use a different stick mode... It will obvously have to be changed in under the `Radio Setup menu`.
But if `Select Model` is entered again. It will default to the Stick Mode that was saved for that model.

**@Ishems** Are you aware that `Stick Mode` select already exists in Open TX?

[image]

Jetrell, 2021-06-22:
To elaborate more on this topic.

I first started flying fixedwings 30years ago on Stick Mode 1. Later in 2005 I took up flying 3D Helicopters. 
Cyclic stirring maneuvers like  _Piro-Flips_ and _Piro-tic tocs_ are much easier to accomplish when the Elevator and Aileron are on the same gimbal. So I chose to use Mode 3. 
Because Mode 2 would require the elevator/throttle spring to be changed as well as the software Stick mode, when going from flying a Mode 1 Fixedwing to a Mode 2 Heli.

Back then I use JR radio's. They did (20years ago) what I am now requesting here.

But I do have flight buddies that fly quads on Stick mode 1. That's why I don't want to solely dedicate my transmitters use, to my Stick mode selection, by doing so in the OI mixing....So they can use my gear, by changing the `Stick mode` over in the `Radio Setup menu`.
But because I'm the Primary user of this TX. This feature would be useful to me, so I don't forget to change between Stick mode 1 and 3, as I go between Fixed wings and MR/Heli's.

lshems, 2021-06-22:
Same for throttle reverse.
Same for switch assignments left and right handed for DLG launch
Same for flaps up or down for slider up or down.
Same for ....

Just copy the model, and change the two input sources.

???

lshems, 2021-06-22:
So you can, just not by mode, but by switching sources on two input lines.

That's what input lines are for.

lshems, 2021-06-22:
Off course you could also do that by modes on model level. But nobody is interested in that but you.

It's not because something CAN be solved in a certain way, and IS solved in a certain way on comparable products, that it SHOULD be solved in the same way.

On other radios there is NO (other) way to do that than by switching MODE per model.

mpjf01, 2021-06-22:
> But I do have flight buddies that fly quads on Stick Mode 1. That's why I don't want to solely accomplish Stick Mode mixing via the Input / Outputs ... I would rather do it via `Stick mode selection` in the `Radio Setup menu`.
> 
> But because I'm the Primary user of this TX. This feature would be useful to me, so I don't forget to change between Stick mode 1 and 3, as I change between flying Fixed wings and MR/Heli's.
> 
> I'm only asking that this transmitter preference be saved to the model memory, as many others are.

What would what you want done accomplish for you - that is what change would you expect to see in the operation of your transmitter compared to the present situation? 

I have no experience with the JR radios that you mention so I don't know what you are expecting as an outcome.

Jetrell, 2021-06-22:
I'm fully aware of how to use work-arounds in OpenTX to accomplish many things. And it wouldn't be such an issue if it was only a couple of models. But when these changes have to be made to dozens of models in the memory, and duplicate them for the sake of two slightly different conditions... 

Remembering what the variation differences between each model in the memory is for, will become far more complicated than just saving the stick mode with that specific model, or changing the stick mode in the radio setup menu.

The person who implemented Stick Mode selection in OpenTX, obviously thought it was a worthwhile feature to have. 
This feature request is simple - **For the Stick Mode to be saved to the individual models memory, as many other preferences are**.

>
> 
> 
> Off course you could also do that by modes on model level. But nobody is interested in that but you.

Don't be surprised how many people would find such a feature useful. The vast majority of open source users never come to Github to ask.

Anyway.. I'm not in any hurry. I'll leave this open. maybe some one else will find the need one day and add it.

@mpjf01 Thanks for your comments
@lshems Thanks for your forceful, over opinionated, condescending and unhelpful comments.

lshems, 2021-06-22:
Your welcome. Next time, if you want the death penalty introduced because other countries use it so effectively, I'll join the discussion again.

mpjf01, 2021-06-22:
> I'm fully aware of how to use work-arounds in OpenTX to accomplish many things. And it wouldn't be such an issue if it was only a couple of models. But when these changes have to be made to dozens [...]

I was genuinely interested in your answer to my previous question as I am sure you would not be making your request here without good reason.  I don't understand your expectations simply because I don't have your experience with your previous transmitters.

Jetrell, 2021-06-23:
>This would mean that OpenTx would have to ask the user to select the mode in the wizard while using the wizard to create a new model. Is that all you are looking for?

@mpjf01  I explained it to the best I can in post 1 and 18. 

Most radio's come default with Stick Mode 1 or 2.
It can already be selected in OpenTX. And many user's change it after their purchase, to suit the `Stick Mode` they fly with. And that might be the only time they change it.
While other user's like myself, use two different Stick Modes, according to the type of model we fly... Heli or Fixed wing.
The goal of this feature, is to prevent the user from forgetting to change the `stick mode`, after they enter `Select Model`.
Which is only relevant to people that fly with two different stick modes.

Whenever  `Backup Model` is entered, under the `Model Selection page.` The `Stick mode` would also be saved with those model parameters.  If the Stick mode is always the same.. as is for most people, Then it would always save the same Stick mode with all models.

mpjf01, 2021-06-23:
I guess the thinking in the design was that changing the stick mode in most use actual cases involves hardware modifications and people won't be doing those on a model by model basis. 

Moving the mode from a global attribute to a model attribute might involve a fair amount of effort (depending on the data model used). Might be something that the EDGE devs could incorporate more easily.

In the meantime, if you haven't already done this, you could include an arming routine that will not work if the wrong mode is selected.

bigtwin, 2021-06-23:
@Jetrell - Why dont you use to different transmitters - one for _Mode1/Fixed Wings_ and one for _Mode3/Copters?_ May this even introduce some more advantages in individual handling of your "mode1" and "mode3"  type models.
I never, ever met somebody who actively was using 2 different stick modes or was even talking about this.

Jetrell, 2021-06-23:
> I never, ever met somebody who actively was using 2 different stick modes or was even talking about this.

Its more common than you may think. Especially with old school pilots like myself, who learned to fly on IC engines back in the 90's.  And move to helicopters and multi-rotors along the way.

fabo137, 2021-06-24:
@Jetrell  

So honestly speaking like all the other guys here it doesn't really matter what kind of "Stick Mode" you have in your settings. Once you have created you model, you map your inputs to your mixes as you want to have. And this setting is stored you "your" model. What would be the benefit of having a stick mode attribute on a model level, it's not even used on that level. 

Other old radios maybe not have the options to flexibly map input and outputs as you want, and therefore "stick mode change" is required. But not on open tx. You have your inputs and you map to your outputs, based on your preferences on the specific model.
 
What you are mentioning as a "work around" is actually a feature giving you the option to map everything however you want. You even can put your Elevator on your poti. Nice not? ;)

Stickmode helps maybe in faster setup in open tx as you have a default setting for new models, but nothing more. 

I really don't get the point what you want to have. Sorry. :(

Jetrell, 2021-06-24:
@fabo137  I am aware of how OpenTX works.
The Stick mode selection can be left in Mode 1. With stick Inputs mixed to the Outputs. For that mode.
* And for Mode 3, mix Roll to Yaw and Yaw  to Roll. Then set _Multiplex_ to  _Replace_ on both.
Or do the same with Mode 2 to Mode 4.
_Then I can double up on every model in the memory and name each one according to the Stick mode it uses._

closed 2021-06-24 by Jetrell