opentx/opentx #6353: Strange behavior of multiple mixer rows with "slow" attribute.

prikrylm, 2019-03-03:
Scenario:
Two mixes on channel 1 something like:

```
 CH1  SB Weight(+100%) Flight mode(FM0:1) Slow(u3:d3) [Flaps]
   += SB Weight(+100%) Flight mode(FM1:2) Slow(u2:d2) [Flaps]
```
when SB is not in middle position, switching of mode (FM1-2-1) produce strange channel output. 
Both channels has its "multiplex" attribute set to ADD. 
It looks like result for channel is computed from active and inactive rows.

When row, in mixer definition, becomes inactive stores some value (somewhere) as value for time when row is activated again.
In case of REPLACE (multiplex attribute) store actual source value, in case of ADD it is zero.
(it looks like heritage from earlier versions..)

This stored value leads to strange channel behavior.

Solution should be using current channel value instead of this stored value, when mixer row became active. In case of mixer rows of ADD type add only active rows. 

The advantage may be that if you do not need to calculate the value of the slow mixer line if it is inactive, you save some CPU time.

kilrah, 2019-03-04:
It's how it is. You should not use more than one line with a Slow on any channel, use high channels for combining them.

prikrylm, 2019-03-04:
I know it is a bit theoretical, but even if I use only one slowed row in channel, behavior of channel output is "strange".  

```
CH1 SA Weight(+100%) Flight mode(FM0:1) Slow(u2:d2)
 += SA Weight(+100%) Flight mode(FM1:2)
```
(Set SA to + or - 100% and switch modes)

kilrah, 2019-03-04:
Yeah. Only put a single line on the channel then.

It's basically impossible to change anything in there without completely breaking compatibility, so for the foreseeable future you just have to deal with it.

prikrylm, 2019-03-04:
OK, I never used such mixing for flying. It was for "servo tester". So it is not necessary to change it, especially if  it is so difficult to change this behavior.

Squirlier, 2019-04-08:
possible workaround:
The Slow attribute does only work within that mixing where it was setup.
one input with some magic settings and an output.
it slows only that input change to the output.
another mixing row would also do its job wit slowing down but the dont know each other.

so a possible workaround ist to setup those mixing rows to output without slowing down.
the output could be to maybe an unused channel (maybe 15) (lots of them usualy free)
then another mixing row where this channel (15) is the input and the output is the final channel (3)
there you could use the slow down.. it is of course always slowing down the final mixing chain.

I hope i could help with my bad english

kilrah, 2019-04-08:
Yes that's the standard way to deal with it and that I mentioned by "use high channels for combining them".

Squirlier, 2019-04-08:
Oh... now i understand...  
THX

closed 2019-10-06 by kilrah