opentx/opentx #6354: X10 will not eject from Finder

nickbraithwaite, 2019-03-04:
**Describe the bug**
At the end of an Open-Tx session I eject the Horus SD card and attempt to eject X10 but nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Finder'
2. Right Click on 'X10'
3. Scroll down to 'Eject "X10"'
4. Click on Eject X10 - nothing happens

**Expected behavior**
X10 to be ejected

**General information**
 - OpenTX version 2.2.3Jan 7 2019 on Apple Mac
 - Hardware Horus X10s
 - OpenTX Companion

kilrah, 2019-03-05:
Yeah same on all radios, just ignore and disconnect for now.

nickbraithwaite, 2019-03-05:
Hi Andre,

Thank you very much!  Sets my mind at rest

Nick

lthole, 2019-05-01:
A work-around that seems to work most times is to:
a) close all instances of Finder
b) cmd-click on both X10 and NO NAME to select both drives
c) right-click on either one and select 'Eject'

nickbraithwaite, 2019-05-02:
Eureka!!

Thanks very much - and I’m incredibly impressed: a solution follow up after such a long pause.

Ta!

ATB

Nick

Nicholas Braithwaite
nicholasbraithwaite@me.com
Box 401, Hahndorf
S. Australia 5245

nicholasbraithwaite.com.au

https://www.amazon.co.uk/So-what-does-conductor-music/dp/064697145X/ref=sr_1_1?s=books&ie=UTF8&qid=1515381388&sr=1-1&keywords=nicholas+braithwaite

lthole, 2019-05-02:
I have had a problem at least once where I had not opened Finder at all, but only worked with NO NAME using Companion. In this instance I could not eject either NO NAME nor X10. It seems to help having Finder open, and then going through the sequence I described above.

lthole, 2020-02-15:
Further information: I also run Win 10 in a Parallels VM on the same machine. The drives eject fine from Win 10.

nickbraithwaite, 2020-02-16:
Many thanks Lothar. I’m running OpenTx on the Mac side of my computer. But your work around - exit Finder, select on the desktop, delete - has been working fine for me 

Nick

lthole, 2020-02-16:
Nick, I also primarily use Companion on the macos side. I only use the Win10 side for documenting my manuals. 

Just to clarify - do you delete the drive or eject it?

Lothar

nickbraithwaite, 2020-02-16:
Eject

N

lthole, 2020-02-20:
I may have found a possible solution:
Researching this problem on the net revealed that the problem is most likely due to Spotlight trying to index external drives when they are connected. Therefore Finder will not eject the drive because it is busy being scanned by Spotlight.

This link was the most informative: [https://forums.macrumors.com/threads/disable-spotlight-for-any-external-drive.1693660/](url)

I navigated to my X10SD drive (renamed from NO NAME) with Terminal, and created a file called '.metadata_never_index' in the root folder.
192-168-1-8:~ lothar$ cd /Volumes/X10SD 
192-168-1-8:X10SD lothar$ touch .metadata_never_index
192-168-1-8:X10SD lothar$ 

If I open the drive in Finder with hidden files showing, the file is there.

I can now eject the X10SD drive, even with Finder open, as long as Finder is not open on the drive itself.

I have not been game to try this on the 'X10' radio drive, this is up to the devs to work out! At least I now have peace of mind when ejecting the SD Card drive.

So my proposal is that the SD Card contents is made available with this metadata file installed, and hopefully on the X10 drive as well.

lthole, 2021-01-09:
I have found this problem to be almost non-existent on the Taranis X9D+ 2019. The drives eject as expected.

In addition, it is worse on the X10S since 2.3.10.

bigtwin, 2021-05-16:
I accidently stumbled across this. At what point is this, since 2.3.12 is about to be published? I am using Companion 2.3.11 on my MAC OSX Mojave 10.14.6 and never had this problem with my QX7. 
Does the problem occure with all versions of MAC OSX? And since I am interested to get me a X10 ... is it with all Horus or only with X10(s)? 
Thx for more information about this issue!

lthole, 2021-05-16:
On my X10S I have found the '.metadata_never_index' solution described above to work. 
Otherwise this works most times:
A work-around that seems to work most times is to:
a) close all instances of Finder
b) cmd-click on both X10 and NO NAME to select both drives
c) right-click on either one and select 'Eject'

You can also use the Clean Eject app and just ignore the messages if the drives reconnect.

bigtwin, 2021-05-17:
So, if I read between the lines ...
...  it is currently unclear whether this has an dependency to the version of MAC OSX, right? I started with _High Sierra_ and OTX 2.2 about 2 years ago. Meanwhile I am on the versions given above and never had this problem.
... I ever used a QX7. Therefor, it is probably highly related to Horus x10(s)

I have excluded the external SDCards from _Spotlight_. And "My SDCard" is renamed from "No Name" as I dont like spaces in such technical names.

raphaelcoeffic, 2021-05-17:
I think this is not directly related to the bootloader but basically our the mass storage driver implementation (from ST). The same happens when I use it with radio running the main software and MSD is chosen from the menu.

lthole, 2021-05-17:
I have problems disconnecting my X7S, but I have not implemented the  '.metadata_never_index' solution. The easiest one is my X9D+2019.

@bigtwin: out of interest, how did you exclude your SD cards from Spotlight?

bigtwin, 2021-05-17:
System Preferences - Spotlight - Data Protection - "+"
When your SDs are mounted simply choose them to be not considered by Spotlight.

However, as far as I know externals should be excluded by default, but I dont know whether this behavior can be changed/might has changed. For sure I remember that I had this behavior in the past: external filesystems did not want to unmount as spotlight was indexing them. Excluding them from Spotlight, led to an unmount right away.

Also you have to consider that FAT32 is a slow filesystem. Sometime the unmount takes very long as data is still be written.

bigtwin, 2021-12-10:
Meanwhile I have a _Horus 10s_ with _OTX 2.3.14_ and it is managed by according _Companion_ on MAC OSX 10.14.6 (Mac Book Pro Early 2015)

The issue is pretty extreme. I could not properly unmount the Horus' storage several attempts in a row. The hints from @lthole did not help. Neither adding the storage to the exclusion list of Spotlight.

**Observations**:
- It seems like the Finder got somehow hung. I dont have any option to "_Force Unmount_"
- While trying to unmount the FrSky drives, the _Disc Utility_ does not finish to collect drive information as it probably depends on some Finder functionality
- If I simply unplug the USB cable, there is no "Drive not correct ejected" message from MAC OSX. So it seems like the drives are somehow "logically" unmounted.
- I changed the USB cable and could repeat the testcase - the subsequent unmount was working fine. 

I will continue and check whether there is a dependency to the cable.

bigtwin, 2021-12-13:
No, I could not confirm a dependency to the USB cable. Sometimes this cable work, sometime a different one and vice versa. Sometimes the volumes do not show up at all, sometimes no proper unmount in _Finder_. One cable seems to give better connectivity than the other one. As I am only purchasing a "USB-Cable", I have no idea what I should look for to get a "good" cable.

As it worked way more reliable and reproducable with my Taranis QX7, I would suspect a hardware issue when connecting a Horus to a MAC. I dont think that OTX can change here something. However, it is nasty but surely no showstopper and I can live with it.

bigtwin, 2023-10-18:
The issue does not occur with EdgeTX. All fine for me now.