opentx/opentx #6355: Looking for contact for public comment about Nirvana support

joshuabardwell, 2019-03-04:
I'm looking for an OpenTX dev to comment about the timeline for support for the Nirvana radio, as well as the future of OpenTX support on non-FrSky radios. Please contact me at joshuabardwell@gmail.com. Thank you.

3djc, 2019-03-05:
We are comited to NV14, and we see no reason for that to change. As far as timeline are concerned, we do never communicate about those since the work is done by a (way too limited) set of dev who do it on their spare time.

We are interested in supporting new radio (any brand) as long as they bring two things : value added for the customer/market (ie no clone of existing radios) and help (read developer time, money doesn't help here) to do the work. We have even an, information page about it : http://www.open-tx.org/newradios.html.

If you are refering to one of the team member been hired by FrSky, it is accurate : they were not happy with the time we could allocate to their radio, basicaly we could not keep with the rate of change/new product they desired, so it is their way to create time for that. He will work part of his time for FrSky internal stuff, and the rest for OpenTX. That's a way they have found to create more ressources for OpenTX.

3djc, 2019-03-05:
I'm also closing this. We need to focus on dev, not anything else

closed 2019-03-05 by 3djc

joshuabardwell, 2019-03-11:
Thank you for your time in answering. 

Do you have any comment about the decision to implement native R9 support, but not native Crossfire support?

qba667, 2019-03-11:
@joshuabardwell I have added native support for configuration of crossfire to NV14. What more would you like to see?

kilrah, 2019-03-11:
What "decision"? None of us is aware of any such thing.

qba667, 2019-03-11:
I have had first version as well, indeed it has problems with powering of module. But in next one that is currently sold they changes transistors and it seems to work fine. After fixing battery calibration, is is reliable. I was using it during last weekend for few hours on Micro crossfire set to fixed power 100mW.

kilrah, 2019-03-11:
100mW was never really a problem, it's higher powers that start getting iffy. But obviously people try to run them and we can't blame them since their modules are capable of it and they should be able to use them to their full potential.

From my measurements on the first units the external module regulator would supply 5V, and to a max of about 300mA, above that the voltage would fall and eventually it could shut the whole radio down. AFAIK they then reworked boards to change a resistor which boosts the idle voltage to 6.5V so it gets into the allowable range for more modules, but the regulator and upstream supply is the same so power delivery capability is about the same, so max current would be about 230mA now (might be wrong, never received new hardware nor instructions to apply the exact mod they did in prod). 

When asked about external module supply based on experience with a decade's worth of modules we said 7.5V and 1A, at the very least 500mA, which is 2.5-5x more power available. Need that for 1W LRS systems. But that would have meant beefy regulators and current paths there was no more room for at the time we got hardware and could run tests. Or switching the battery config to 2s (there's a reason why everybody with a JR slot does that), which I can only assume was deemed not OK because UFPV had done so much advertising over the useless "battery redundancy" thing and switching one while running another during flight (yeah right, like you've got time for that) which you technically certainly shouldn't even do in the first place since the cells are in parallel with no current limit.

Lots of frustration with this product and its project management on our side. That's the main reason for delays, and somewhat reduced priority.

I fully understand that from the outside some people are thinking "they're sold to FrSky" and putting that forward as a reason for lack the lack of diversity - but that's not correct. 

Yes there is a bias, but that bias is only based on the fact that FrSky:
- Do their best to design original products and features
- Listen to us, many popular features of their radios have come from our vision and desires for products that we wanted to have for ourselves, and hoped a large portion of the market would - such a collaboration is the main thing that mattered to us and we had been unsuccessfully looking for a partner who'd have the guts to get into this for a while unsuccessfully including with European companies - seems it worked out pretty well in the end for those who dared waling outside of the paved walkway
- Trust us, they have literally put their company's stakes in our hands multiple times
- Understand us, there have been frictions at times, but even as a Chinese company with their culture that is absolutely against that have reverted some of their decisions and forgiven us for what they could have seen as unreasonable outbursts
- Respect us, they have always considered our work valuable and made it known
- Have done all of that for 7 years now
- Recognise both our project and us as people enough to hire the original project creator full time under Western rules, with one of his main jobs being to continue contributing to OpenTX, even when they know some competitors who rip them off are going to benefit from it

And finally... have been the only ones to do even just one of all of the above points so far even with the numerous contacts we've had.

We would have, and would still welcome anyone else who came with the same attitude, but that's never happened. Anyone who complains about the lack of competition"just" has to do as well as them. It's "easy".