opentx/opentx #6356: Request to add a new telemetry switch data type to the SPORT protocol to support 32 Telemetry switches

AnthonyPeate, 2019-03-05:
The open TX tea are doing great work thanks.

I have had this idea for some time now and finally though I should submit it as  a suggested enhancement.

The proposed change:
I would like to be able to send a 32 bit telemetry value on the SPORT bus that is mapped in the OpenTX software to 32 telemetry switch values. These could then be used as a new a new switch type in the special functions page to trigger events in exactly the same way that the existing logical or physical switches do.

While I have been using arduino nano to send data to the SPORT using the A3 data type with discrete arbitrary voltage levels to trigger logical switches this is very messy.

My use case for this is to trigger play announcement wave files when servo plugs are not plugged in. But I think this could be very useful sending digital it states for many other applications where the desired outcome is to trigger a corresponding special function based on the telemetry switch state.

Not sure if these telemetry type IDs have been used yet but this is the idea:-
#define TLSW_FIRST_ID 0C00
#define TLSW_LAST_ID 0C00

The 32 bit telemetry value send on the SPORT bus for this new data type would be maps two 32 new Telemetry switch values as follows:-
Bit 0  --> T1 
Bit 1  --> T2
  .      .        .
  .      .        .
Bit 31  --> T32

These new switch types should be available for use in the custom functions and logical switch pages
in the same way that logical switches L1 to L32 and physical switches can be used.

This then allows digital events detected on a model aircraft to trigger special functions.

kilrah, 2019-03-05:
I believe all of this can be done with a Lua script.

3djc, 2019-03-05:
I'm not too fan of the telem decoding side which can easily be done in LUA with a lot of flexibility, but I like the idea of having virtual switches driven through LUA. The only downside is that LUA is not available in emergency mode, which kinda limit the scope of what you can do with it

AnthonyPeate, 2019-03-06:
Looks like i might need to learn how to program LUA scripting to do this. I am use to C/C++ but LUA is new to me.

However I still think this is a great enhancement idea for the SPORT and open TX to support a new telemetry driven switch data type.

Thanks for responding to my request.

closed 2019-03-06 by AnthonyPeate