opentx/opentx #6360: Companion 2.2.3, radio simulator

urkadurka, 2019-03-10:
**Describe the bug**
On Radio simulator window clicking on arrow up/down left/right trim buttons the trim behavior isn't reproduced, instead moving the slider the behavior is ok.

**To Reproduce**
1. Prepare a new model from zero, like an airplane, no modifications will be done.
2. Open the radio simulator
3. Click on UP/Down trim arrow button for elevator or throttle, Left/Right for rudder or aileron 
4. The trim behavior isn't reproduced
5. Instead if you move the slider the trim behavior is reproduced

**Expected behavior**
Clicking on arrow button the expectation is to reproduce the trim behavior

* On The radio the behavior is ok

**General information**
 - OpenTX Companion  2.2.3 on Windows
.  - OpenTX version 2.2.3 on X9D+
- Frsky X9D

3djc, 2019-03-10:
I'm not sure I understand what you are saying. Are you saying clicking in the arrow highlited bellow does not work for you ?

[image]

Because here it definitively does

urkadurka, 2019-03-10:
You are right ! 
By clicking the trim on the simulator doesn't work I would add that the slider doesn't move.
Thanks

3djc, 2019-03-10:
I would suggest you have a local issue, because it does work fine for everybody else

urkadurka, 2019-03-10:
mmm, interesting, this could be fine and preferable.
What do you suggest me to to ?
Thanks

kilrah, 2019-03-10:
Given your other issue it seems you probably have an "adjsut GVAR" SF active. Need to disable that before you can use the keys to trim. Same on radio.

closed 2019-09-06 by kilrah