opentx/opentx #6361: Companion 2.2.3 GVAR controlled by TrmT, TrmE, TrmR, TrmA

urkadurka, 2019-03-10:
Describe the bug
On Radio simulator window moving the slider for a Trim a controlled / linked GVAR don't change, instead using the arrow buttons it works

To Reproduce
Prepare a new model from zero, like an airplane, in special functions create a ADJUST GV1 (or any other) with source TrmT or any other , enable the function and place "ON" on switch column

Open the radio simulator
Move the slider available for the trim change the GV1 doesn't change.
Click on UP/Down trim arrow button the GV1 changes.

Expected behavior
Moving the slider available for the trim the expectation is that the GVAR changes its value

On The radio the behavior is ok

General information
OpenTX Companion 2.2.3 on Windows
. - OpenTX version 2.2.3 on X9D+
Frsky X9D

kilrah, 2019-03-10:
That's as intended. It's the whole reason the buttons were added in the first place.

closed 2019-03-10 by kilrah

urkadurka, 2019-03-10:
Please let me leave a comment, why the trim on simulated radio display changes accordingly to the slider ? Why the trim on simulated radio clicking on arrow button doesn't changes ?
Here two files with images and the saved file form PC Companion
[Trim.pdf](https://github.com/opentx/opentx/files/2949590/Trim.pdf)
[Trim.zip](https://github.com/opentx/opentx/files/2949591/Trim.zip)

Hoping in your answer
Thanks

kilrah, 2019-03-10:
As said above, when you have an Adjust GVAR function with a trim as source, that trim's keys serve to adjust the GVAR and NOT to trim.
The slider in the simulator however accesses the trim directly so continues to work.

3djc, 2019-03-10:
[Simulateur Radio (FrSky Taranis X9D+) - Phase de Vol 0 (#0) 2019-03-10 15-56-12.zip](https://github.com/opentx/opentx/files/2949600/Simulateur.Radio.FrSky.Taranis.X9D%2B.-.Phase.de.Vol.0.0.2019-03-10.15-56-12.zip)

Have a look at that, you'll see it works fine unless you have a adjust GVAR function or something else in your setup preventing it

urkadurka, 2019-03-10:
Thanks 3djc !
If I disable the GVAR adjusting by TrmT my simulator works fine too !
For my understanding why using a trim to modify a GVAR this behavior in present ?
This my better knowledge.
Thanks

3djc, 2019-03-10:
Because when you use that feature you turn trim into gvar adjust, therefore it is NOT trim anymore

urkadurka, 2019-03-10:
Thanks, that's clear and logic .
I'm satisfied 👍 
Thanks