opentx/opentx #6364: Taranis 9+ and companion2.2.3 Jan 7 2018

AndyCapp77, 2019-03-13:
Unable to add a model image to the model in both Companion and the TX model setup

3djc, 2019-03-13:
Likely you image doesn't have the right format/size. Please test with this picture to confirm it

[Alpha.zip](https://github.com/opentx/opentx/files/2960190/Alpha.zip)

lepicfoto, 2019-03-14:
yup it is the same here. it is not possible in any way to add a picture, it is not possible in companion, and yes all paths are correct and work fine :-) and browse for a picture is also not possible, the same in my tarnis x9D and x9E. so I have to agree, it seems that it is not possible to use pictures tested with your Alpha pic.
BUT it IS possible to use custom splash screen :-)
Jesper.

3djc, 2019-03-14:
[image]

Not an issue here, you path or image name have to be wrong

lepicfoto, 2019-03-18:
Yes yes YEEES it works, only who could know that there is "REALLY VERY FORBIDDEN" to use a logical sorting of images and use subfolders and such, just as you normally do on computers, even stupid process controllers have a folder order. Not in this case, just collect ALL your pictures in the IMAGES folder, and then it works just fine GGGGG. Just making fun, it's NOT in bad sense whatsoever.
I am mega humble, with utmost respect. It works just fine and it gets so much better all the time. with the utmost respect.
Jesper malling
Denmark

3djc, 2019-03-18:
Well, there is kindof a hint : we had to reduce filenames to 6 chars (10 including the extension) for memory/eeprom reason, where would we store path with just 6 chars ? But yes, you are definitively right, no subfolders

closed 2019-03-18 by 3djc