opentx/opentx #6370: Compilation 2.2 problem

RanBiron, 2019-03-14:
I made fresh installation then I followed very strictly instruction on the page : https://github.com/opentx/opentx/wiki/OpenTX-2.2-Linux-Build-Instructions,

OK Results：
1、cmake -DPCB=X9D+ -DGVARS=YES -DLUA=YES -DDEBUG=YES -DCMAKE_BUILD_TYPE=Debug ../
2、make firmware
3、make libsimulator
Not OK Resutls :
    make companion22：
/home/book/opentx/companion/src/modeledit/inputs.cpp: In member function ‘void InputsPanel::AddInputLine(int)’:
/home/book/opentx/companion/src/modeledit/inputs.cpp:112:22: error: ‘const class QVector<const ExpoData*>’ has no member named ‘constFirst’
     newChan = (expos.constFirst() == &md);
                      ^
/home/book/opentx/companion/src/modeledit/inputs.cpp:113:22: error: ‘const class QVector<const ExpoData*>’ has no member named ‘constLast’
     hasSibs = (expos.constLast() != &md);
                      ^
companion/src/modeledit/CMakeFiles/modeledit.dir/build.make:226: recipe for target 'companion/src/modeledit/CMakeFiles/modeledit.dir/inputs.cpp.o' failed
make[3]: *** [companion/src/modeledit/CMakeFiles/modeledit.dir/inputs.cpp.o] Error 1
CMakeFiles/Makefile2:2092: recipe for target 'companion/src/modeledit/CMakeFiles/modeledit.dir/all' failed
make[2]: *** [companion/src/modeledit/CMakeFiles/modeledit.dir/all] Error 2
CMakeFiles/Makefile2:1899: recipe for target 'companion/src/CMakeFiles/companion22.dir/rule' failed
make[1]: *** [companion/src/CMakeFiles/companion22.dir/rule] Error 2
Makefile:769: recipe for target 'companion22' failed
make: *** [companion22] Error 2

I have not modified the source code, why is there an error?

kilrah, 2019-03-14:
The Q7 version in your distro is too old.

closed 2019-03-14 by kilrah

RanBiron, 2019-03-25:
> The Qt version in your distro is too old.

> The Qt version in your distro is too old.

May I ask what version I need at the minimum?

kilrah, 2019-03-25:
We build with 5.7.1, but I believe the minimum is 5.6.