opentx/opentx #6373: Port to DIY ESP32 based boards.

tmrttmrt, 2019-03-22:
As my winter hobby project I work on port of OpenTX to ESP32. At the moment I have a basic working version of the radio firmware. I still need to figure out how to integrate it to Companion.  The branch is at: https://github.com/tmrttmrt/opentx/tree/esp32/ if anyone is interested.

Maybe it should be pulled into the upstream tree as a separate branch so others can test whether it breaks compilation for ARM. I wonder whether there would be enough interest and DIY enthusiasts to support the version.

kilrah, 2019-03-23:
Interesting, the ESP32 has low I/O count, so how are you dealing with all the inputs? External processor? What are you using the ESP32's unique features for?

tmrttmrt, 2019-03-23:
The pin count is indeed low. Two MCP23017 - i2c 16 input/output port expanders are a simple solution.

The main advantage of ESP32 for DIY is that is easy accessible and cheap with well established development environment.

My plan is to use the built-in WiFi for connection to Companion as well as non-latency-critical RC links eliminating the need for external modules. There are already some projects for this that could be incorporated.  LoRa boards are another possibility for a compact RC link.

renebohne, 2019-03-31:
I just had a look into your code. Great job, thanks! I tried to get low level access to the radio of the esp32 some time ago because I hoped that it can be used for general 2.4Ghz transmission. But no luck so far. If it was possible, there would be no need for an external nRF24L01 if somebody wants to do Bayang, MJX, or SymaX... 
I will get a ST7565P display and try your code soon. Did not look too deep, but are there the SPI pins that you use for the display? 
#define PIN_NUM_MOSI GPIO_NUM_12
#define PIN_NUM_CLK  GPIO_NUM_14
#define PIN_NUM_CS   GPIO_NUM_13

tmrttmrt, 2019-04-01:
I doubt that the radio part is completely software defined and emulating any protocol is possible.  There exists, however, the esp32 long range WiFi mode that might be useful for RC.

The SPI pins in esp32 are quite flexible. At the moment my code uses all the pins so one can only swap pins to avoid any collision.  You also need at least two additional auxiliary pins to drive LCD:

#define PIN_NUM_DC   GPIO_NUM_27 // This pin selects the LCD command or data registers.
#define PIN_NUM_RST  GPIO_NUM_0 // Reset, it might be done by an external circuit on power on

spectrenoir06, 2019-04-21:
I need to try this on my esp32 radio http://antoine.doussaud.org/esp32_radio
what is working right know ?

kilrah, 2019-04-21:
This is very nice! Lack of switches is a little disturbing though!

spectrenoir06, 2019-04-21:
Yeah it's was hard decision but the radio have to triggers buttons ( enough for arm + mode )
but the radio have a MCP3008 with 8 ADC input available

tmrttmrt, 2019-04-22:
> I need to try this on my esp32 radio http://antoine.doussaud.org/esp32_radio
> what is working right know ?

I had noticed your radio before I started. Nice work!

Please see https://github.com/tmrttmrt/opentx/blob/esp32/radio/src/targets/common/esp32/Readme.md for the current status. Only PPM output is implemented now.  I am about to start some field tests. Before I do  I would like to implement SD card removal/insertion handling.

powermik, 2019-04-25:
Would be awesome to support esp32 modules with integrated oled

powermik, 2019-04-25:
can you post a quick schematic on how to connect the pots/switches?

tmrttmrt, 2019-04-25:
https://github.com/tmrttmrt/opentx/blob/esp32/doc/esp-wroom-32/esp-wroom-32.pdf

tmrttmrt, 2019-05-18:
The code has been pushed to 2.3 version by removing the MEGA2650 parts and basing everything on  the SKY9X-board code.

The ESP32 board was also included in Companion.

tmrttmrt, 2019-06-02:
ESP-NOW protocol implemented as an internal module. Briefly tested with ESP8266 as a receiver on a micro quad.

The receiver code is at https://github.com/tmrttmrt/opentx-espnow-rx.

tmrttmrt, 2019-06-11:
Ported onto a board with OLED display.

Zebble, 2020-01-02:
For the life of me I can't seem to get this to build.  I'm using the 4.0 release of esp-idf on Windows, installed with the esp-idf tool and installed mingw32 stuff.  Cmake seems to work but no "flash" option gets added to the makefile so the flash command doesn't work...  Any pointers on creating a build environment on Windows that I might have missed?

tmrttmrt, 2020-01-02:
Nothing special to do. I am using esp-idf 4.0-dev snapshot. I start mingw as this:
`set IDF_PATH=/some-path/esp-idf 
T:\win64app\msys64\msys2_shell.cmd -here -mingw32
`

Do esp-idf examples compile and flash as expected?

Zebble, 2020-01-02:
It looks like esp-idf release 4.0 (or higher) no longer use Msys2.  It all runs from a standard windows command prompt.
The examples compile and flash perfectly.  Just need to figure out how to use idf.py to do the compile and flash.  Not having much luck so far.

tmrttmrt, 2020-01-05:
I updated the build files to post v4.0 IDF (branch esp32-idf-4.0). See readme.md under [...]/esp32 dir how to build. The ota upload target is unfortunately still broken in this branch.

naorunaoru, 2020-01-21:
So I tried to build it on macOS and failed. `esp32-2.3` branch fails to build because of mingw32 dependency, `esp32-idf-4.0` fails because it couldn't find partition table. `esp-idf` version on my machine is 4.1, maybe that's not helping too.

My C skills are kinda crap so I'll try again on my Windows machine in a couple days.

Thank you for this project though, ESP32 is a fun little thing and I'd definitely choose it over AVR even if it requires using external port expanders.

Zebble, 2020-01-21:
Had the same issues as you @naorunaoru.  I got a bit further but could never get the bin to compile so I gave up.  The instructions (and updated instructions) point to still using msys tools to compile, which esp-idf 4.0+ don't officially support anymore.  Any attempt I've made to get things working through Windows or Linux have ended up with the same issue...

I gave up for now.

naorunaoru, 2020-01-21:
Sad news. @tmrttmrt please help with debugging these issues.

naorunaoru, 2020-01-21:
Installed fresh mingw, added latest xtensa-esp32-elf toolchain, shuffled partitions.csv around a lot.

Now I'm here. No idf.includes generated indeed.

```
Scanning dependencies of target firmware
[ 37%] Building CXX object radio/src/CMakeFiles/firmware.dir/debug.cpp.obj
xtensa-esp32-elf-g++.exe: error: C:/MinGW/msys/1.0/home/naoru/opentx/build/idf.includes: No such file or directory
```

tmrttmrt, 2020-01-21:
> So I tried to build it on macOS and failed. `esp32-2.3` branch fails to build because of MingW dependency:
> 

esp32-2.3 branch is outdated. Use the esp32 branch with esp-idf 4.0-dev git snapshot.

naorunaoru, 2020-01-21:
> > So I tried to build it on macOS and failed. `esp32-2.3` branch fails to build because of MingW dependency:
> 
> esp32-2.3 branch is outdated. Use the esp32 branch with esp-idf 4.0-dev git snapshot.

Forgot to mention, I used esp-idf 4.0-rc on Windows last time I tried with esp32-idf-4.0 branch of your fork. I'll try to use 4.0-dev later.

tmrttmrt, 2020-01-21:
Something got broken during upstream merging after tag 2.3.4-esp32 and a clean build does not work. Please try this tag with  esp-idf release/v4.0.  I can perform clean build on Windows.

tmrttmrt, 2020-01-22:
The tip of esp32-idf-4.0 branch compiles with esp-idf commit 39f090a4f1dee4e325f8109d880bf3627034d839 (I do not know how have I got this one checked out?) On my machine it is marked as origin/4.0 in SourceTree, but I cannot find it on GitHub. Sorry for 
confusion.

Zebble, 2020-01-22:
Thanks @tmrttmrt.

I'm still confused.  Are you saying we need to download a specific commit of esp-idf and/or this repo?  Are you still preferring the legacy esp-idf install (gnu make/msys2/mingw) over the newer esp-idf tools installed (cmake/Windows console)?

Zebble, 2020-01-22:
> Installed fresh mingw, added latest xtensa-esp32-elf toolchain, shuffled partitions.csv around a lot.
>
> Now I'm here. No idf.includes generated indeed.
>
> ```
> Scanning dependencies of target firmware
> [ [...]

I'm in exactly the same spot now, so there's that...   :)

tmrttmrt, 2020-01-22:
I tried to build on Ubuntu 16.04 and another Windows box and realized that the idf build process is much less robust than I assumed. Currently it builds only on my particular windows box with certain version of tools. I will now try to find how to reproduce the working build system on Windows and then start to work on compilation in the esp-idf v4.1 release.

Zebble, 2020-01-22:
That's a good way to describe it tmrttmrt.  Definitely not very robust and things drastically change/break between releases.  Seems to be getting better with 4.x though...
Looking forward to see how you create your build system.

naorunaoru, 2020-01-22:
`[100%] Built target flash`

I did it.

xtensa-esp32-elf toolchain version 5.2.0
tmrttmrt/opentx HEAD at d2b2754fa
esp-idf HEAD at tag v3.3.1 (what, why) with all submodules reinited

I don't have any peripherals connected to ESP32, even display, so I can't verify if it compiled successfully, but that's something.

tmrttmrt, 2020-01-23:
> I tried to build on Ubuntu 16.04 and another Windows box and realized that the idf build process is much less robust than I assumed. 

I had broken esp-idf installation on Ubuntu box. Following the instruction for esp-idf v3.3.1 to setup the tool-chain it builds straight out of the box as noted by naorunaoru.

In windows it builds as well under MSYS2 MINGW32 shell provided together with esp-idf v3.3.1 toolchain. The MSYS2 environments, however, needs some tweaking:
Cmake for windows must be installed separately and included into MSYS2 path.
` pacman -S  mingw-w64-x86_64-make
pacman -S mingw-w64-i686-python2-pillow
export PATH=/mingw64/bin:$PATH
`

naorunaoru, 2020-01-23:
> Updating the MSYS2 environment seems to break things so avoid it.

I've got the latest MingW with MSYS2. The only thing I did a bit differently is that I added `-DCMAKE_SH="CMAKE_SH-NOTFOUND"` to `cmake` because I use MSYS2 shell instead of cmd/powershell.

tmrttmrt, 2020-01-25:
I  fixed esp32-idf-4.0 branch to compile with esp-idf v4.1 in Windows and Linux. 
In Windows MSYS2 environment is still required, but I had to use the native Windows python and cmake. 

I also had to significantly modify the PPM driver since RMT library was changed in v4.1. There might be issues due to this.

naorunaoru, 2020-01-26:
I can confirm that it now builds on macOS. Super happy to use my usual development machine. 
Still waiting for the multiprotocol module to arrive. Already tested on a breadboard with gimbals, switches and buttons — it required a bit of specific tweaking here and there, but it works!

@tmrttmrt beware, if this port will ever be merged to upstream there will be tons and tons of inexpensive RC controllers with ESP32 and OpenTX. :)

naorunaoru, 2020-02-03:
@tmrttmrt please advise on pullup resistors for buttons. I sometimes get weird misclicks on buttons connected to the MCPs.

tmrttmrt, 2020-02-05:
Please look at the schematics. 100 nF capacitors parallel to the inputs and 100-200 Ohm resistors in series to the switches should do. Put both close to the inputs to filter out also any RF picked up by wires.

Zebble, 2020-02-06:
@naorunaoru here's the link to the PDF schematic in case you can't find it:
[https://github.com/tmrttmrt/opentx/blob/esp32-idf-4.0/doc/esp-wroom-32/esp32-wroom-32.pdf](url)

I still can't get a build environment to work.  If anyone has the energy to create a build-doc for one it would be greatly appreciated.  Windows or Linux, I don't care.

tmrttmrt, 2020-02-07:
On linux shuld be quite straigthforward. Install the he esp-idf as specified at https://docs.espressif.com/projects/esp-idf/en/latest/get-started/ . 

```
. ~/[path-to-esp-idf]/export.sh
python -m pip install Pillow
cd [src-path]
git clone -b esp32-idf-4.0 https://github.com/tmrttmrt/opentx.git
mkdir build
cd build
cmake  ../opentx/radio/src/targets/common/esp32/esp-idf  -DPCB=ESP_WROOM_32  -DCMAKE_TOOLCHAIN_FILE=$IDF_PATH/tools/cmake/toolchain-esp32.cmake
make flash ESPPORT=/dev/[tty-connected-to-esp32-board]
```

Zebble, 2020-02-07:
Well, that worked!  Probably was obviously in my cmake line.  Now to build up a circuit!
Thanks @tmrttmrt

Zebble, 2020-02-07:
> Ported to a board with an SSD1306 OLED display.

Any pointers on how to build for SSD1306 support?

tmrttmrt, 2020-02-08:
-DPCB=ESP_HELTEC_32 configures for the Heltec  SSD1306 OLED board (https://heltec.org/project/wifi-kit-32/).  This configuration uses only one MCP23017 limiting the number of switches.

The pin definitions are in [opentx-src-folder]\radio\src\targets\[board]\board.h:

```
#define I2C_DISPLAY_SCL_GPIO          GPIO_NUM_15
#define I2C_DISPLAY_SDA_GPIO          GPIO_NUM_4
#define I2C_DISPLAY_RESET_GPIO        GPIO_NUM_16
```

When porting to other boards one should also take a look at CONFIG_ESP32_XTAL_FREQ in   [opentx-src-folder]\radio\src\targets\[board]\esp-idf\sdkconfig.defaults to select proper xtal frequency for the particular board.

alexandrecnunes, 2020-05-19:
Is there another link to the schematic? Because this is not ok.
https://github.com/tmrttmrt/opentx/blob/esp32-idf-4.0/doc/esp-wroom-32/esp32-wroom-32.pdf

tmrttmrt, 2020-05-19:
Please be more specific about what is not OK?

alexandrecnunes, 2020-05-19:
excuse me,
when I try to access the link, it does not lead to a correct page. 
But now,I got it
Thank you

gbraad, 2020-08-13:
@tmrttmrt Has there been any development/enhancements in recent time?
@spectrenoir06 interesting project, and exactly what I am trying to achieve. the size is great as that is perfect for small hands (like kids), unfortunately I wasn't able to find any schenmatics or design files.

I have some ESP32 modules that were sent to me by mistake (wrong part) and think this is ideal for a small transmitter that can even operate without a display (configured over bluetooth by a phone or webbrowser). At the moment more interested in getting it to work with the nRF24 from the MPM, but adding the serial protocol would certainly be great.

tmrttmrt, 2020-08-13:
> Has there been any development/enhancements in recent time?
Unfortunately not.

FractalEngineer, 2020-10-25:
@tmrttmrt Hey mate awesome work with this

I'd love to experiment with it making a modular TX; what's the current status on it; I saw in the schematics that PPM and power management is not yet implemented; is it still the case?

Does that mean that it can't yet work with external TX modules?

tmrttmrt, 2020-10-26:
The git version is the last version. 
It can work with external modules that accept the PPM input since the PPM output is active. The software-controlled power hold and external PPM signal routing are not implemented.

pierreh, 2020-11-01:
@tmrttmrt thanks for the great work!
I've managed to build this and flash it to a base ESP32-S, which is like the ESP32-WROVER.
I have to disable the `per10msTask` because it fails to communicate over I2C.
I don't need a lot of switches and no trims so I would like to use it without the IO expander and I want to implement the serial driver for an external module.
Can you give some pointers as to how to do those two tasks?

tmrttmrt, 2020-11-02:
per10msTask should not be disabled since it performs quite a few important things in OpenTX.  Look into "keys_driver.cpp" for disabling I2C reads. 
For the serial driver I cannot help.

pierreh, 2020-11-10:
Got it working without IO expander, with an oled display and sd card. Really cool! I've got some IO expanders on the way though, because I can only get 3 switch positions now. Just a few pins short ;-)
Enabling SPORT for an external module looks doable to me, but it may take a while to figure out all the details.
I've tried to get Lua support to build, but I'm running in to issues with the IDF cmake.
Have you tried to build Lua support?

tmrttmrt, 2020-11-10:
Nice to have it working. I have not tried to build Lua. Unfortunately I cannot help much with cmake since my knowledge of cmake is on the level of guessing.

I have noticed however, that when building with  esp-idf versions above v4.0 the sound output and SD card are in some kind of conflict and either the sound output or SD card do not work.

closed 2021-09-04 by stale[bot]

bangdc90, 2021-09-29:
@tmrttmrt 
Thank you for amazing project.
I am trying to bootup GLCD ST7565 with your sour code, but I am facing a problem that after mainGui show, it disappear immediatly. I tried many config for sequence of ST7565 but still fail.
Do you know any reason cause that issue?

bangdc90, 2021-10-02:
I made it work after comment initAudio(); in board.cpp
And also need to disable reading/writing I2C in key_driver.cpp because i dont use external expand chip

tmrttmrt, 2021-10-02:
Great! I have not touched the project for a while and I am currently too busy with my regular job to be able to help you. The absence of external chips would definitely hang the program.  initAudio() should, however, not interfere with the display.

royyandzakiy, 2021-11-27:
Hi, I think it's a little bit late, but I have just stumbled upon this great project. I've seen that this has the option to use PPM to connect to external modules. But is it able to connect using the CRSF protocol, so that I could connect a ELRS module using the TX pins?

Thanks in advance!

tmrttmrt, 2021-11-28:
Unfortunately not.

geofrancis, 2021-12-01:
have you seen espnow? it might be a viable way of doing a long range rc link with these. 
https://discuss.ardupilot.org/t/simple-esp-telemetry-serial-bridges/75536

tmrttmrt, 2021-12-02:
I have already implemented it, so one does not need any external module.  The receiver part (https://github.com/tmrttmrt/opentx-espnow-rx) is still quite rudimentary.

[11 more replies]