opentx/opentx #6374: Taranis E voltage indicator bar

frobert0, 2019-03-23:
With my FrSky Taranis E it comes to a false display when Spannugsanzeige bars in the upper display which manifests itself as follows.

The second green bar is always there even if the first yellow or less bar is no longer displayed.

Is someone familiar with this mistake and where does it come from?

Robert

kilrah, 2019-03-23:
Never seen that before, probably a faulty segment.

Or maybe the fact that the operation timer has far exceeded max capacity, maybe reset it.

frobert0, 2019-03-23:
> Never seen that before, probably a faulty segment.
> 
> Or maybe the fact that the operation timer has far exceeded max capacity, maybe reset it.

Reset operating timer has helped.
Thank you

kilrah, 2019-03-24:
I believe the timer capacity was increased at some point, so you're probably on an old OpenTX version.

closed 2019-03-24 by kilrah

frobert0, 2019-03-24:
OpenTX version is 2.2.3

frobert0, 2019-03-24:
> I believe the timer capacity was increased at some point, so you're probably on an old OpenTX version.

OpenTX version is 2.2.3

3djc, 2019-03-24:
A fix has been pushed to futures 2.2.4 / 2.3 to prevent this in the future