opentx/opentx #6378: bug in zchar2str(char * dest, const char * src, int size)

tmrttmrt, 2019-03-24:
In strhelpers.cpp: 
```
int zchar2str(char * dest, const char * src, int size)
{
  for (int c=0; c<size; c++) {
    dest[c] = idx2char(src[c]);
  }
  do {
    dest[size--] = '\0';
  } while (size >= 0 && dest[size] == ' ');
  return size+1;
}
```
`dest[size--] = '\0';` maximum index for char [] is size-1.
It should be `dest[--size] = '\0';`.

philmoz, 2019-04-07:
I'm new to this project; but the existing code looks correct to me.

It looks wrong if you assume that 'src' is a zero terminated C string (the lack of comments doesn't help).

Looking at the usage in the rest of the code I believe that the 'src' parameter is not a C style zero terminated string - it is a char array of length 'size'.

The 'dest' parameter is a char array of length 'size' + 1 (or more).

tmrttmrt, 2019-04-07:
I agree.

closed 2019-04-07 by tmrttmrt