opentx/opentx #6379: Companion. Conversion of model from X9D to X10S reverses direction of LS.

mpjf01, 2019-03-24:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

mpjf01, 2019-03-24:
Models using the LS on my X9D radio get the direction of operation of the LS reversed when Companion converts from the X9D to the X10S. The RS direction is the same as it was on the X9D. There were warnings for the models concerned but those just asked to confirm the selection of S2.

kilrah, 2019-03-25:
OpenTX has no setting for slider direction, you're just seeing the result of different hardware.

mpjf01, 2019-03-25:
I guess I expected the conversion process to alert me to the fact that the direction was reversed - just as it does for other hardware differences.

kilrah, 2019-03-25:
The warnings are based on availability or not of a choice on a given board. We have no place to store whether that choice that is named the same does the same on various models.

That's why you need to carefully check everything on any conversion.

closed 2019-08-19 by kilrah