opentx/opentx #6381: Companion 2.2.3 Print Model Outputs: erratic behavior

Miami-Mike, 2019-03-26:
It appears to me that Companion 2.2.3 "stable" has an erratic behavior where _Print Model_ and _Compare Models_ will show some channels in Outputs that haven't been changed from their default settings, and **not** show some channels that **have** been changed.

It seems to happen only with models created with that version, because models that were created with earlier versions of 2.2 are displaying okay for me.

kilrah, 2019-03-27:
Can you supply an example file?

Miami-Mike, 2019-03-29:
Never mind, I just realized how it works. In Print Model and Compare Models, a channel is displayed on the Outputs page if it has a corresponding mix on the Mixes page, which makes good sense now that I think about it.

Sorry about that.

closed 2019-03-29 by Miami-Mike