opentx/opentx #6382: 9XR Pro - Expect ver: 2.2ver0018

Missoulaoblongata, 2019-03-31:
Hello,

So I've recently went through flashing a Turnigy 9XR Pro with OpenTX 2.2.3 and everything went pretty smoothly.  I used the download function in companion to get the latest version of firmware and then used the download button to get the latest sdcard file which I copied.  It took me to this page

https://downloads.open-tx.org/2.2/release/sdcard/opentx-9xrpro/ 

which appears to be correct but the latest file is sdcard-9xarm-2.2V0013.zip which is not 2.2V0018.  Is the radio wrong in thinking it needs 2.2V0018 or is there a missing sdcard file?

Thanks

kilrah, 2019-04-01:
Looks like the process to generate that hasn't been working for a while. Just get the pack for a Taranis, it just has extra stuff that is unused on a 9XR-Pro but what is used is the same.

Missoulaoblongata, 2019-04-02:
> 
> 
> Looks like the process to generate that hasn't been working for a while. Just get the pack for a Taranis, it just has extra stuff that is unused on a 9XR-Pro but what is used is the same.

Thanks for the suggestion I'll give it a shot and see how it goes.  It looks like all the taranis sdcard files are the same so I'll just go with the one for the x9d.

3djc, 2019-04-02:
The files have been manually generated here : https://downloads.open-tx.org/2.2/release/sdcard/opentx-9xrpro/

closed 2019-04-02 by 3djc