opentx/opentx #6388: Enabling frsky telemetry when using external RF module while disabling internal

yanivasy, 2019-04-11:
The external bay includes a PPM/SBUS output to enable control by an external RF module.
Telemetry such as the fantastic yapuu script is accomplished by the SPORT protocol.
The external bay includes also an input/output pin called SPORT which is common with the internal module.
I've connected the FC telemetry via an SPORT adaptor and it is able to decode frsky telemetry data only when Taranis is bind to X8R receiver in D16 mode (although the receiver doesn't send any data).
When disabling the internal module, no telemetry is displayed.
The feature requested is enable telemetry when internal RF module is off.

3djc, 2019-04-11:
Telemetry is enabled when internal module is off, not sure what you are doing

yanivasy, 2019-04-11:
I set the external module to output PPM for controlling.
In D16 mode of the internal module, while the SPORT pin of the external module is connected to FC telemetry port with SPORT adaptor, telemetry data is displayed on Taranis.
In this case, the X8R receiver doesn't send any telemetry data.
When disabling the internal RF module (or changing to another mode), telemetry data is not display.

3djc, 2019-04-11:
What external module are you using ?

yanivasy, 2019-04-11:
I'm trying to use RFD900x data link which has both PPM passthrough for control and UART (57600) data link for telemetry. Most users use this telemetry channel for the ground station. I'm trying to use it for yappu script.

3djc, 2019-04-11:
Have you set your telem as FrSky SPORT in telem screen ?

yanivasy, 2019-04-11:
I'm currently holding the Taranis in my hand. Looking at Telemetry page (12/13) starting with RSSI setup.
Maybe I'm missing something but I can't see any FrSky SPORT settings for telemetry.

3djc, 2019-04-11:
You don't have something like : 

[image]

yanivasy, 2019-04-11:
No. Maybe I need to set a specific build option for enabling it?
Or maybe downloading a specific/newer  OpenTx version ?

yanivasy, 2019-04-11:
Sorry, when dissabling the internal RF module, telemetry type can be change.
However, it is set to FrSky S.PORT

kilrah, 2019-04-11:
That data link is almost certainly not giving out sport telemetry.

You didn't say what flight controller you were using and how it was set up.

yanivasy, 2019-04-11:
I'm planning to use RFD900x data link but havn't connected all wires yet. I've tested the PPM control functionality and it working.
I'm using Pixhawk 2 (the cube). Telemetry is working by using X8R receiver that I want to replace with RFD 900 x.
For the perposse of testing, I connected the FC's telemetry port (TELEM2) which is set to mode 10 (FrSky passthrough) by wires (without using the RF link) and SPORT adaptor to the Taranis SPORT pin within the external RF bay.
In this setup, telemetry was working only while the internal RF was set to D12 and bind to the X8R receiver. Needless to say that the X8R receiver's SPORT was disconnected from the FC.
Can't see why such configuration won't work.

kilrah, 2019-04-11:
It's perfectly normal. 
In the SPORT system the module / receiver is the communication master, and is the one asking devices "do you have something to tell me".

Your FC is supposed to be connected to a receiver, which is the one managing that communication. If you connect it straight to the radio, since the radio doesn't ask anything you will have no data. The radio is only listening to the communication that the installed module would be managing.

kilrah, 2019-04-11:
Note that your testing is not at all representative of the setup you'll have with the RFD900 in the first place.

closed 2019-04-11 by kilrah

yanivasy, 2019-04-11:
Just to make sure that I see you point, does the Taranis send any questions/requestes to the receiver?
If so, it stops from sending them while no receiver is connected.

kilrah, 2019-04-11:
The Taranis does not send anything. That is the module/receiver's job.

Only when a module and receiver are communicating will you get requests . These requests are intended to be answered on the receiver side, but since they are also carried over to the ground side it "kinda works" to answer them there, although that will cause conflicts if something is answering the same request on the receiver side.

Basically it's not meant to be used that way at all. The device in the link I gave you handles that all correctly for you.

yanivasy, 2019-04-11:
Thanks for the "Mavlink To FrSky Passthrough Converter" link.
However, this requires an additional hardware for doing some simple software conversion.
By Taranis, I meant both the Taranis controller and its internal module.
To clear this issue, please confirm that the internal RF module (within the Taranis) sends requests to the receiver.

kilrah, 2019-04-11:
No, the module does not send requsts to the receiver. The receiver sends requests itself.

yanivasy, 2019-04-11:
You got me all confused:
If the receiver sends requests itself, why can't Taranis send these exact same requests to the FC and get replies?
SPORT is using a 57600 baud rate - the same as the suggested data link and as MAVLINK.
The conversion from UART telemetry to SPORT can be done easily after the data link before connecting to the SPORT pin of Taranis.
According to my suggestion, the additional hardware will be a simple dual inverter (RS232-TTL) and a diode. This will not include any software.
I truly can't understand why it is not possible?

schwabe, 2019-04-11:
@yanivasy Opentx *could* do that. But there no reason to this normally. Normally the XJT and the receiver handle all this. From Sport protocol perspective the TX is more like a sensor than anything else.

The use cases of connecting sensors directly to the TX are quite limited so nobody bothred to emulate a receiver in OpenTX to query sensors.

On top of that S.Port is a single master bus and the master is normally either the receiver or TX module. Adding a second mastery is asking for trouble.

schwabe, 2019-04-11:
It would be easier to modify the FC software to just send the sensor data without being queried by a master. That way OpenTX would pick up the sensor data.

yanivasy, 2019-04-11:
@schwabe,
I totally agree with you that modifying the FC software to just send data without being queried is the best way to go. That way only one way data link is required - the downlink (with uplink free for other usages).
As for adding a second master, my suggestion included removing the first master and using only one at a time.
Thanks.

kilrah, 2019-04-11:
> According to my suggestion, the additional hardware will be a simple dual inverter (RS232-TTL) and a diode. This will not include any software.
I truly can't understand why it is not possible?

Sport is timing-dependent, it will not work corectly if the ground sends the requests and the FC answers, the delay will mess everything up and cause collisions. When a request comes the device has 4ms to start sending its answer.

So either use the solution I described, or modify the FC software to send data on its own.

yanivasy, 2019-04-11:
Issued:
https://github.com/ArduPilot/ardupilot/issues/11084

Hartinger, 2022-10-01:
I have found this old issue, which is dealing with the very proplem I have.

I have programmed a converter that transforms telemetry data from an M-Link module into the FrSky D-Format, for injection into my Taranis with OpenTx 2.3.15 via the corresponding module bay pin. This has worked perfectly for over 5 years and now I have modified the converter such that it provides the telemetry data in the S-Port format (without the need for any polling, as this is not provided by the radio).

I have a setup with just two voltage values being sent to the radio, but sensor discovery does not find any sensor. Now I wonder if I have misunderstood something, and the Taranis is not able to accept and process S-Port data from the module bay pin. Or are there any specific requirements that I have ignored, e.g. an RSSI data packet must be present in order that the radio accepts the telemetry data.

Hartinger, 2022-10-03:
Based on further tests, it seems that OpenTx needs to see also the polling packets in order to accept and display the sensor telemetry data.. If only the sensor packets are present, sensor discovery does not find anything.

Can anyone confirm this?