opentx/opentx #6391: Channel latches with input using MAX and LS

lthole, 2019-04-14:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

lthole, 2019-04-14:
If an Input with MAX as source and an LS as switch is mixed to a channel, the channel will latch high.

Example:
LS01 
AND SA↓ SB↓

IP05
Source = MAX
Switch = LS01

CH05
Source = IP05

When both SA and SB are down, CH05 will go to 100%. When either or both switches are switched up, CH05 stays at 100%.

kilrah, 2019-04-15:
Perfectly normal. An input with no active line maintains its value.

closed 2019-04-15 by kilrah