opentx/opentx #6396: Telemetry shows 0 values in telemetry page

bepr, 2019-04-17:
**Describe the bug**
After upgrade from 2.1.6 to 2.2.3 the telemetry cell voltages and current show 0 in the configuration and telemetry page. The main screen top bar however does show the lowest cell voltage (3.76V) correctly. The configuration page show that we collecting data from the FrSky telemetry by showing blinking asterisks.
Note the "cels" unit needed to be set to "V" after the upgrade.

**To Reproduce**
Steps to see the telemetry screen:
1. Go to main screen
2. Long push "page" button
3. Cell and current values show "0"

**Expected behavior**
Cell and Current values should show real values.

**General information**
 - OpenTX version 2.2.3
 - Hardware Frsky X9D+
 - OpenTX Companion 2.2.3 on Windows 10

**Screenshots**

[image]
[image]
[image]

kilrah, 2019-04-18:
Delete the Cels sensor, rediscover it, reselect it where needed.

bepr, 2019-04-18:
Hi Andre,

I did delete all sensors and did a rediscover with no avail. Even tried to fix it with building a new model from scratch. Reinstalled firmware and bootloader all with no avail.

I should have mentioned that. My bad.

bepr, 2019-04-18:
I see this thread was closed. However issue is not solved. Fact is there is communication as I can see cel voltage in the top bar of the main screen.

Per your advise I did delete and re-discovered the sensors again.

 

1)deleted all sensors

2) discovered all sensors

3) stop discovery

 

Here are the screenshots:

 

4) edit Cels

Description: cid:AA316FB2-C03D-458C-B8FB-00A43272BFFC

Ongoing discovery. You can see the Cels and Curr show up with a blinking asterisk.

 

Description: cid:FC633A60-F729-4E1B-8DC6-D020D444BFD6

Edit screen of Cels right after discovery. 

 

Closed #6396 <https://github.com/opentx/opentx/issues/6396> .

bepr, 2019-04-18:
[image]
Ongoing discovery. You can see the Cels and Curr show up with a blinking asterisk

[image]
Edit screen of Cels right after discovery.

3djc, 2019-04-18:
The ticket is closed because there is plenty of evidence that OpenTx works fine with FLVSS sensors, so no bug on that front. If you experience issue (and we are clearly not saying you don't), then it is a local implementation issue, which this place is not meant to handle.

Please go to community places, like rcgroup oir openrcforums for end user support

bepr, 2019-04-18:
It worked fine with version 2.1.6 and 2.1.9 But understood.

Thanks for the advice. Will do.

bepr, 2019-04-18:
Allow me to add 1 more finding. The log shows all voltages, cels, cel1, cel2 and cel3 in this case. It’s difinetly collecting data. Just does not show it.

kilrah, 2019-04-18:
Is it a real FrSky FLVSS or a third party sensor?

bepr, 2019-04-18:
FrSky FLVSS with a X8R receiver flashed to latest firmware. Also tried a FLVSS without success.

Just that data does come in and are logged. Also as stated before I can see correct readouts in the top bar of the main display. A mystery why it’s not showing in the telemetry screens.

kilrah, 2019-04-18:
Can you try connecting just the FLVSS and no other sensor and make a new model/discover?

bepr, 2019-04-18:
I tried that too. Just that before adding something to Github you try all what’s possible first.

 

So far using 2.1.6 all ok.

Flash to 2.1.9 still ok.

Then flashed with 2.2.3 (firmware and bootloader) using the 2.2 SD card contents from the web and created a new model. No avail.

Strangest is besides the main screen top bar, the log shows the voltages as well.

Reverting back to 2.1.6 and all works fine again.

Mystery.

kilrah, 2019-04-18:
> Just that before adding something to Github you try all what’s possible first.

Well most people don't, so we tend to expect nothing that hasn't been explicitly mentioned.

I have no idea. Can you post your otx file and a log? Maybe we need to make you a test firmware with raw sport log.

bepr, 2019-04-18:
Thanks for reopening.
Per your request:
Today's log 
[Sbach_342-2019-04-18.xlsx](https://github.com/opentx/opentx/files/3095643/Sbach_342-2019-04-18.xlsx)

The OTX file

[GitHub-2019-04-18-2.2.3.otx.zip](https://github.com/opentx/opentx/files/3095667/GitHub-2019-04-18-2.2.3.otx.zip)

bepr, 2019-04-18:
Sorry, still finding my way. Reopen again.

bepr, 2019-04-18:
What I did not do is a radio factory reset, flash with 2.2.3 again, format EEPROM and build a model from scratch. Please let me know if you want to do that.

kilrah, 2019-04-18:
You have the Cels sensor set as unit V. That is not correct. Heve you edited it after discovery?

bepr, 2019-04-18:
As it bothers me, just did a factory reset anyway. Steps all using the radio:

1)      Backup EEPROM

2)      Factory Reset (EEPROM formatting)

3)      Performed calibration

4)      Modified radio settings to reflect mode 2

5)      Edit model 01 to get the channel mix right

6)      Discovered sensors

 

No luck, telemetry screen show successful communication with lipo and current sensor but values stay zero

Now am out of possibilities to try. As am using the voltage to tell me when to land, am obligated to revert back to 2.1 soon.

More than glad to get you whatever you think could help here. Just let me know.

bepr, 2019-04-18:
Could have set to "V" as I tried different things. Attached is the OTX file with no telemetry edit made after the factory reset. No "V" here.
[NewModel-2019-04-18-2.2.3.otx.zip](https://github.com/opentx/opentx/files/3095870/NewModel-2019-04-18-2.2.3.otx.zip)

kilrah, 2019-04-19:
What is the full filename of the 2.2.3 firmware you have flashed?

bepr, 2019-04-19:
On the radio display

Vers: 2.2.3 (ae5ac05b)

 

From the download page:

Opentx-x9d+-faimode-noheli-lua-luac-en-2.2.3

3djc, 2019-04-19:
Then the explanation is clear : faimode prevent all telemetry but a few selected item

bepr, 2019-04-19:
My goodness. That was it. What a relief.

FAI mode is supposed to disable all telemetry displays, other than RSSI and voltage, to comply with Federation Aeronautique Internationale (FAI) regulations. If voltage means the radio battery voltage then we found a backdoor as the main display top bar does shows Telemetry voltage. If voltage means all voltages then the question pops up why only the top bar.

3djc, 2019-04-19:
No, it was the result of design choices, more details here : https://github.com/opentx/opentx/issues/3443

closed 2019-04-19 by 3djc