opentx/opentx #6399: Display Stick and uSec position on Curve graph

Jetrell, 2019-04-20:
Currently, there seems to be no way to overlay the stick and micro second position on the curve graph.

[image]

Even my old Spektrum and JR radio's can do this.  

DX6 display
[image]

Its very helpful when setting up a throttle curve, if you can see exactly were about's the stick is positioned with the Input and Output uSec values. So the curve can be plotted more precisely and quickly.

The Inputs page has a similar feature. but it doesn't relate to the Curves page.

[image]

It makes it very tedious when trying to fine tune any curve. If we have to keep jumping back and forth between the Curve and Channel monitor screens.

lshems, 2019-05-11:
For info: is fixed by me and others using Lua.

Eldenroot, 2019-10-03:
Good idea nad I miss this from Spektrum radios

ripper-911, 2020-02-04:
Glad I found this as I was just about to submit it as a feature request. Would be the bomb for matching servos using the curve functions.

Eldenroot, 2020-08-02:
PR is ready :)

PR #7707 (2020-09-22): Add cursor to custom curves - Closes #6399
Closes #6399

Adds a cursor and in/out values to custom curves (as in edit input menu) when editing from edit input menu or edit mix menu
commit 869f8127 (2020-09-22): Add cursor to custom curves - Closes #6399 (#7707)

Add cursor to custom curves
Refactor functions declaration
Cosmetics
closed 2020-09-22 by bsongis