opentx/opentx #6402: NEG/POS in Inputs unexpected behaviour

Allerhopp, 2019-04-27:
OpenTX 2.2.3 X9D and Companion

Using POS and NEG in Inputs leaves a random result, if the stick is in the opposite half. Using curves instead POS/NEG gives the expected result.

OTX File attached
[Neg_Pos.zip](https://github.com/opentx/opentx/files/3124101/Neg_Pos.zip)

kilrah, 2019-04-27:
POS deactivates the line whenever the source is negative and vice versa.
An input without an active line keeps its last value.

Your job to do things right to get what you need given the above definitions. It is usually recommended to keep a default line at the bottom of each input should all others get deactivated unless you're sure you understand how it works, common advice given on forums.

closed 2019-04-27 by kilrah

Allerhopp, 2019-04-27:
I expected the source to become zero before it turns negative. But i will contact forums for advice.