opentx/opentx #6403: OpenTx build setup fails for flight tested binaries as per Docker file.

Melbong, 2019-04-30:
Hello,

I would like to know the correct way to reproduce the production build system of OpenTx.
As per the [Wiki](https://github.com/opentx/opentx/wiki/OpenTX-2.2-Linux-Build-Instructions) the flight tested binaries are only from the build server and a [Docker file ](https://github.com/opentx/opentx/blob/2.2/radio/util/Dockerfile) is provided for it.

But trying the commands from docker file I see that it is either not updated or not used any more.
For example - 
$ wget -q http://ftp.de.debian.org/debian/pool/main/g/gcc-avr/gcc-avr_4.7.2-2_amd64.deb fails as there is no such file on the debian repository.

The travis.yml at the root of project uses entirely different build set-up, OS and compilers.

I tried setting up build system(FW, companions and simu) on Ubuntu 16.04 as well, following the instructions on the Wiki.
Again only able to build using the Qt-Creator(because of higher version of Qt).
The build for 16.04 also fails if using Cmake CLI or IDE based, as the default Qt version in the 16.04 repo is 5.6 and Opentx branch 2.2 needs Qt 5.7 atleast.

I'm trying to match my build environment with the one used by OpenTx for releases so that I can use the flight tested binaries as per the wiki.

kilrah, 2019-04-30:
You should be able to use the Dockerfile with a modern system.
Looks like they have thrown out gcc-avr the version we used, so some updates are needed. 
But since we dropped AVR support in 2.2 we don't need it anymore anyway, can be removed.

Melbong, 2019-04-30:
Hello kilrah,

Appreciate the quick response.

These are the packages related to AVR which don't need to be downloaded any more -
_gcc-avr_4.7.2-2_amd64.deb_, _avr-libc_1.8.0-2_all.deb_ and _binutils-avr_2.24+Atmel3.4.4-1_amd64.deb_.
I will try to remove them and build again.

Also regarding the OpenTx release is image created by the Docker File or Travis CI used(They are both very different using different OS-Debian Stretch vs Ubuntu Trusty, AVR compiler versions-not used I know but still)?
While setting up my own build environment I would prefer to be as close to the release set-up as possible.

3djc, 2019-04-30:
Travis does offer limited flexibility in term of available enviroment, so we pick what is closest for that specfic purpose of compilation checks. We build firmware based on the docker file.

Melbong, 2019-04-30:
Hello 3djc,

Thanks a lot for the confirmation.

kilrah, 2019-04-30:
Travis is just for "does it build or is there a compilation error on a platform I didn't check for", its output is never used.

Melbong, 2019-05-02:
Thanks Kilrah for the confirmation as well.

So, I removed the above mentioned AVR related packages but the version of _libmpc2_0.9-4_amd64.deb_ also seems to be updated in Debian Stretch. 
I couldn't figured out if this library is used although considering it is math functions it might as well be.

kilrah, 2019-05-02:
https://github.com/opentx/opentx/commit/02e56d245d46b09a64422fc27c6ebcf90ece2412

Melbong, 2019-05-02:
Ohh didn't see that commit.
Great thanks.

Melbong, 2019-05-02:
Hello,

I'm able to build after making the required changes.
Thanks.

closed 2019-05-02 by Melbong