opentx/opentx #6404: Crossfire not working on X-Lite pro

salem83, 2019-04-30:
I just received my X-lite pro.
Unfortunately, The CF lite module isn't working.
1, CRSF is not an option for the external RF module
2. when selecting PPM, DSM2 or SBUS, the module shuts down within 3 seconds.

Hardware: X-Lite Pro
Firmware 2.3.0 xlites (05f0057b)

Is this intentional or just not implemented yet?

kilrah, 2019-04-30:
It will be there once our builds are available. The factory firmware has the options FrSky decides to enable.

salem83, 2019-05-01:
Thanks. 
Any estimate about a release date?

3djc, 2019-05-01:
Sorry, but there is no such thing as ETA with us.

LUM-TECH, 2019-05-09:
[image]

Says the guy whose last commit to this repository was 2 years ago. Nice!

3djc, 2019-05-09:
Hum, why don't you look at OpenTx repository, not a fork I'm not using ?

LUM-TECH, 2019-05-09:
Oh my gosh, you're right... you are a huge contributor.

kilrah, 2019-05-09:
You must still be looking at the wrong place.

LUM-TECH, 2019-05-09:
No, I was saying that he was right.

LUM-TECH, 2019-05-10:
Why is there no such thing as an ETA? I'm confused.

3djc, 2019-05-10:
Because we do that in addition of our day job. When you know you'll be in the office say 8 hours/day, you can plan somewhat. When you don't have that view, and do best effort in your free time, it is not realistic to plan

LUM-TECH, 2019-05-10:
I see. Makes sense. Well, keep up the good work.

kilrah, 2019-05-10:
https://github.com/opentx/opentx/commits/2.3

Look and learn instead of sh%&/(ing on people who do the work.

backman8571, 2019-05-10:
So is it 3 fork's ??

FrSky, Flysky ( NV-14) and Jumper TX
Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

3djc, 2019-05-10:
The list of forks can be found here: https://github.com/opentx/opentx/network/members

LUM-TECH, 2019-05-10:
No one's shttng on anyone. I think you are mistaking sincerity for sarc. Thanks 3djc.

kilrah, 2019-05-10:
No, it's just that we got to read your comments through notifications even when you deleted them.

LUM-TECH, 2019-05-10:
Lol, you saw the one that I deleted, you mean? Because I was wrong and admitted it?

LUM-TECH, 2019-05-10:
You seem flustered. 3djc is the one that it was directed at, and he calmly replied and I answered back with an admission that I was wrong. And he continued to reply calmly to other comments. I wasn't even talking to you lol..

kalo6, 2019-05-14:
i would love to know more on that issue , proximate time till crossfire will be supported, will it operate at the full speed ?

kilrah, 2019-05-14:
FrSky never include 3rd party protocols in their firmware builds, and it is not known at this point whether OpenTX can technically make it work on the hardware or not.

tgreer, 2019-05-14:
> FrSky never include 3rd party protocols in their firmware builds, and given the newest information from FrSky it is not known at this point whether OpenTX can technically make it work on the hardware or not.

Are you suggesting there may be hardware changes in the x-lite pro which will stop crossfire working?

3djc, 2019-05-14:
https://www.frsky-rc.com/frsky-advanced-communication-control-elevated-spread-spectrum-access-protocol-release/

Q: Regarding the external module for the FrSky new coming radios. new
A:FrSky will mostly adopt the 8 pins module interface (Lite version) for the new coming radios. For the manufacturers of the third-party external module who follow the requirements below, FrSky will offer the necessary supports of software and hardware including the technical material, interface design and etc.
1. The third-party external modules should be verified by FrSky compatibility tests;
2. The manufacturer does not infringe on others (individual, organization and company) intellectual properties.
Using an unverified external module can cause product impairment. The unverified third-party manufacturer is responsible for any damages that may occur, and the FrSky users warranty will be invalid.

Not sure what this all means

salem83, 2019-05-20:
Simply - FrSky decides which modules we can use or not. First line will be their own Lite-Module series.
If a 3rd-party manufacturer wants to be in, they have to comply their rules.

> ... given the newest information from FrSky it is not known at this point whether OpenTX can technically make it work on the hardware or not.

Or they just don't want to (yet) ... 
It had worked on the original X-Lite ... even if wasn't officially supported.
This may have generated to much supported support cases when someone followed the instructions but damaged the radio/module - because "use at sour own risk".
Another thought - they just wanna "keep out" competitors and promote their own stuff like the R9 series. 

I've had the R9M Lite module, bus wasn't fully happy with it so i bought a CF to make some comparison. 

What wonders me, even if i build a firmware from the current 2.3 branch, the CRSF protocol isn't available. I've managed to re-enable it, but the module doesn't turn on when selected. So there may be some more stuff which has been disabled. As well on other protocols like DSMX or SBUS, the module gets shut down after a few seconds. 
Any idea about this?

3djc, 2019-05-20:
Crossfire missing from 2.3 nightly firmware is an issue from us, will investigate and fix it

3djc, 2019-05-20:
Here we go fixed in 2.3 (well in tomorows nightly really). Crossfire is now a companion option  so you can choose it.

Now will the new FrSky radio ever support it remains to be seen. As for your module turning off after a few seconds, your are not the first one mentionning it. Maybe you want to adress that question to FrSky, since we never turn module unless instructed by user in OpenTX

3djc, 2019-05-20:
N10 and later version should support it :)

zimzillakilla, 2019-05-20:
3djc, my r9m lite module turns off by itself as well. i noticed  that ignoring the splash screen warnings triggers this issue 100% of the time. I now make sure all switches are at their default position and the throttle all the way down and wont interrupt the Lady speaking. :) :)

3djc, 2019-05-21:
We have done everything we can on OpenTX side, so I'm closing this, but you need to be aware of that: 
 https://www.rcgroups.com/forums/showpost.php?p=41906517&postcount=162

This will likely prevent it to work

closed 2019-05-21 by 3djc

salem83, 2019-05-22:
Cool thanks! I'll check it out when i've time.