opentx/opentx #6408: Opentx 2.2.3 - stops playing sounds

morel666, 2019-05-03:
**Describe the bug**
Sometimes my 9XR Pro with OpenTx 2.2.3 stops playing sounds. Radio boots normally, plays "welcome to OpenTx" sound, plays warning sounds if applicable, even starts playing sounds assigned to switches etc. After some time of usage all sounds are muted. When radio is restarted everything starts to work properly. 

**To Reproduce**
Use radio with sounds enabled. It happens not always, I can't describe steps that will guarantee the issue is reproduced. Sometimes I can use radio for many minutes without any problem, sometimes problem happens just after a few seconds after start. 

**Expected behavior**
Radio plays sounds all the time.

**General information**
 - OpenTX version 2.2.3
 - Hardware 9XR Pro

kilrah, 2019-05-03:
This is often down to the SD card, format and reload or try another one. Could be the cause of your other issue too.

L4ky, 2019-05-08:
I have the same issue with my 9XR Pro, same firmware. Even old 2.2.2 had same behaviour. Using stock sdcard.
I will try another one

Bogdanu2018, 2019-05-16:
I have same for 1-2 years, I reseted and formatted everything....same ...sounds disappear after few minutes. After restart of tx is working again.  Same radio 9xrPRO. It made me crazy because I have vario sensor and telemetry but I cant hear any alert...soooooo.... the only work around which I found(by mistake) and is really working is to play some sound all time, if some sound or alert is playing all the time all other sounds works too. So what I did is on glider I just activate vario sound on stick and I leave it activated with very low volume, this way all my telemetry alerts works too. Or when I fly other models without vario, I just config them as they have vario and if I leave vario sound on(activated by some stick) BUT without vario sensor connected there is no vario sound but all other sounds keep working all time. I hope you understand if you dont I will explain better. 
In short, what I noticed is that the bug comes when there is some timeout for sound. If there is no alert for few minutes, the timeout activates and stops all sounds but if there is something playing constant, bug disappear!!! Some developer should know better whats happening!

L4ky, 2019-05-26:
I tried a different microsd and the issue persist.
After x minutes, it stops playing sounds.

closed 2020-02-20 by stale[bot]