opentx/opentx #6410: Feature Request: Single Point Curve

Gfyss, 2019-05-03:
Helicopters using governors almost always use a flat throttle curve to set the desired RPM.  OpenTX currently supports curves with 2-17 points.  In order to make helicopter setups easier and less prone to errors, can you please add an option for a 1-point curve?

The desired behavior would be a single output value that remained the same across the entire control range.  On the display you would see a flat line that went up or down depending on the user input.

The current method of using a 2-point curve and setting both end-points to the same value does work, but can be prone to errors if the user is in a hurry or has imperfect vision (I don't always have my reading glasses handy).

kilrah, 2019-05-03:
Makes no sense using a curve then, just use a fixed value (MAX).

closed 2019-05-03 by kilrah

Gfyss, 2019-05-04:
Well, I finally got the MAX function to cooperate, but it doesn't really work for me.  I still prefer the curve option because I can easily switch to a different curve if I need to program or re-calibrate the ESC with the throttle channel.  Curves work for me and I'd rather not re-program my brain.

Thanks for the suggestion though!

kilrah, 2019-05-04:
You can have multiple input lines with MAX source and different weight values that you switch in the exact same way as your lines with different curves, you can have a global variable adjusted by special functions or flight modes... there are literally several ways to do this that are cleaner than a 1-point "fake curve".