opentx/opentx #6415: Lua enhancement idea: Reset-Telemetry/Reset-Flight on Horus Widget

offer-shmuely, 2019-05-08:
Some of the Horus Widget show values with state, like MIN/MAX/Average
We can write new widget that display them in our way, like gauge and graph
but there is **not way to know when to clean the data between flights** except of turning off/on the transmitter
   
Suggested solution:
add to the api of the widget "**reset=reset**"
so we can write a resert() function to clean the old data

function reset()
  myMin = -1
  myMax = -1
  myAvg = -1
end

function create(zone, options)
...
end

function update(wgt, options)
...
end

local function background(wgt)
...
end

local function refresh(wgt)
...
end

return { name="MyWidget", options=_options, create=create, update=update, background=background, refresh=refresh, **reset=reset** }

lshems, 2019-05-08:
You can reset any data whenever you wish. Am I missing something?

Reset telemetry, flight, etc.

offer-shmuely, 2019-05-09:
how do I know in horus widget code, that the user did the following operation?

[image]

or this:

[image]

lshems, 2019-05-09:
If you use the min and max telemetry values, you don't need to. They get updated automatically.

Post the script somewhere and I'll have a look where it's stuck.

offer-shmuely, 2019-05-09:
https://github.com/offer-shmuely/opentx/blob/2.3/radio/sdcard/horus/WIDGETS/BattCheck/main.lua

lshems, 2019-05-10:
Send me a PM on rcgroups with your contact details if you want to discuss it.

I have installed the widget, and checked with some cells, but have no clue on how to intepret the widget data. So I also don't get what the problem is.

I do see that you have the cels as widget setting, but no cels-. So you cannot use the already available cels- values in your widget. I think that the problem is there. You should not try to manage the cels- yourself, but use the predefined sensor cels-. That one is also reset.

teckel12, 2019-05-11:
@offer-shmuely My script (INAV Lua Telemetry) does this by manipulating timer3. It starts timer3 and then monitors when timer3 is reset via the reset telemetry function. When it detects this, it runs it's own reset function.

lshems, 2019-05-11:
Ok, just had a chat with offer-shmuely. about a same kind of workaround: check if cels- is different than your own cels-. Timer3 might be easier.

But as a general solution, you want to now the exact reset type that was triggered, flight, timer, telemetry, ...

We agreed that a nice solution would be to use events.

Why not, for widgets, add a reset event to be passed to the background function and refresh function, where opentx will pass the type of reset if executed, just like a button event is passed in a normal lua telemetry script to the run function?

So expand the event parameter with the reset events, and pass it to any foreground and background function (you need it in the run for widgets, so you can pass it yourself to the background that isn't called by opentx itself in that case. You need it in the background when the foreground is not running.)

For telemetry scripts, where the background is called by opentx, you need it in the background only, but for simplicity, I would include it in the run call as well. 

for widgets, this change could be done in one go with the normal button events pass through that is asked for a long time now. the event forwarded to the refresh could then contain both button as reset events, the to the background only the reset events.

offer-shmuely, 2019-05-11:
I agree with lshems that the reset notification will be match more generic if it will exposed through a general purpose mechanism like “events” that also transfer other info (like rssi-low, user buttons, telemetry-lost, ...)

teckel12, 2019-05-12:
@lshems For sure, a proper method of detecting a reset would be welcome. Key events would be as good a method as any, and technically, a telemetry or flight reset is a key event.

Special care would need to be taken, however. Because a reset happens so quickly a script may miss the event. While the event could also last a longer time, then special care may need to be taken in the script so a reset wouldn't happen several times.

With my timer3 method, these problems don't exist, and I can always catch when a reset happens. Sure, I can't tell if it's a telemetry or flight reset, and timer 3 functionality is lost. But honestly, it works really well and I've never heard of someone using timer 3.

I could give a basic breakdown on how the logic works for using timer3. I had a few rounds before getting it perfected. But it's rock-solid at this point.

lshems, 2019-05-12:
It is no problem, or should not be, to make sure from opentx to just put the event in a cue, and post it to the next call.

That's no rocketscience, but proper design.

The workaround doesn't work for the battery, as people might choose to only reset a timer, and not the telemetry.

It's not for me, but for @offer-shmuely .

I don't use battery sensors. Just trying to help get a proper Auktion instead if complicating widget design.

teckel12, 2019-05-12:
@lshems It's been a request dating back a couple years, so it's a low priority. I haven't spent much time sifting through the OpenTX code, but maybe I should to take care of possibly easy changes like this.

lshems, 2019-11-08:
Please reconsider expanding events with os events like reset flight, etc.

offer-shmuely, 2019-11-09:
I hope you will reconsider to implement such event.
I like to continue and enhance the selection of graphical widget for the horus, and I am missing a method to do so.

closed 2020-08-05 by stale[bot]

vprheli, 2023-08-28:
HI, I am still searching the solution. We need reset out local variables which are calculated on telemetry values for instance. If sensor send current and voltage for instance and we calculate min max power. There is necessary reset these values correctly. Of course in some cases we can compare some sensor varables to estimate that reset telemetry was called. But it is not pure solution.