opentx/opentx #6418: Opentx included widgets use global and local variables outside the widget instance table

lshems, 2019-05-11:
The Lua widgets included on the SDcard are not safe for use together with other widgets, or when several instances of the same widget are used.

There is abundant use of global and widget local variables.

All variables should be part of the first array 
that is sent to the create function, and is further managed by opentx, often named zone.

I have made a pull request on the fork of @offer-shmuely for the battery widget.

I suggest to start checking proper coding in Lua scripts being part of the delivery.

It will get a mess otherwise that is hard to clean.

offer-shmuely, 2019-05-13:
No more global parameters in Batt-Check widget
see pull request 6412

offer-shmuely, 2019-10-01:
is possible to merge issue:6412 and close this issue as well?
or it needs additional changes?

offer-shmuely, 2020-01-26:
please close this bug, it's already fixed on 6412

offer-shmuely, 2020-01-31:
and the updated code on #7338 does NOT include global variable as well.
please close this issue, or comment what is need to be fixed additionaly

closed 2020-02-14 by lshems