opentx/opentx #6420: Need way more inputs / parameters for mixing scripts

azaz44, 2019-05-13:
**Problem**
I'm running OpenTX 2.2.2 on FrSky Horus x10.

Mixing scripts are limited to have 6 inputs per script. This is very limiting and I often have problems with that while writing scripts.

**Requested change**
Ideal solution would be to have this unlimited or limited to some big number like 64 or more.

If, for some reason, it is technically impossible to make so many inputs, it would be great if OpenTX would only limit "SOURCE" inputs to 6, but allow for more VALUE type inputs. This way the script can give user options to set. Please note the difference between SOURCE and VALUE:

```
local input =   {
  { "Is Armed", SOURCE },
  { "Thr On", SOURCE },
  { "Min time", VALUE, 0, 15 * 60, 30 },
}
```

**Workaround**
The workaroud I'm using is very ugly, but I can't find anything else.
I write more than one script. The first script is what really does the work. Other scripts are there just to allow for more inputs, and they set it to global variables, so the first script can use it. For example, 3 scripts allow for 18 inputs this way.

**Example where it is needed**

The script I'm developing now, is for flying RC Helicopters. User is able to specify that he should not perform some maneuvers. For example he does not want to fly upright-ccw-nose-down-funnel. This is because he already knows this maneuver very well and tends to fly it very often instead of learning other 15 variations. The scripts then detects when this maneuver is being performed and reminds user (beeps) to stop it and work on something else. So basically reminds user to fly things which he needs to train more instead of those which he's good at already.

This mixing scripts needs an input parameter for each maneuver, so user can choose if he wants to get a warning when this maneuver is performed.

Number of maneuvers can easily reach 40 or 50. Just circles and funnels happen to have 16 variations.

3djc, 2019-05-13:
We can discuss it for Horus where memory is not an issue (if it ever happens, that would not be before 2.4 at the earliest), but have you considered making an LUA UI for setting those things and save that to a file then used by the other scripts ?

kilrah, 2019-05-13:
You should probably be doing your own settings management. Lua can read/write to files, so you can put all your settings in a file in any way you want.

azaz44, 2019-05-13:
Yeah, configuration script sounds like a good idea. Not sure why I haven't considered that. I already have some UI controls to make that, I will try this way.

> What kind of thing are you doing?

There is one example in issue description.

I also made another script in the past which messes the helicopter every some time in certain ways and user has to recover from this situation. All for learning. It had 12 parameters to configure so I used 2-scripts approach. But this one would also be suitable for having UI configuration over a standalone script.

kilrah, 2019-05-13:
Yep hadn't seen. I think it doesn't even have to be a mixer script then, could have a widget processing it.

azaz44, 2019-05-13:
The script generates an output which tells if user is doing forbidden maneuver or not.  So I think this has to be a mixing script. Then there's a logical switch and a special function connected to it, to play sound / voice / vibrate in this case. It's more flexible this way. Also this allows user to have it enabled / disabled with a switch etc. Could even cut off the throttle as well but that would be too rude maybe ;)

azaz44, 2019-05-13:
Hmm, I reopened the issue.

I just realized, the solution with using external standalone script to configure the thing has one big disadvantage - it is radio specific. I will develop it for Horus, then nobody with Taranis will be able to use it (for example). As opposed to a script which can be run on anything OpenTX.

For this particular script UI is probably the only option because the amount of options is so huge. But I think many other scripts would benefit from having more options there to configure script behavior. It doesn't have to be 64, but something like 16 would be great. 6 is really limiting.

Then, on top of that, developing a UI is complicated. Fine for me as I made some development with that already and can reuse components, plus it is fun to do. But actually it very often means 4h spend on writing great mixing script, and then 4 days spent on making UI for that. Developing UI is just so complex.

So the bottom line is: giving more options (16 maybe) is still a great improvement to OpenTX.

lshems, 2019-08-28:
Check my blog for an easy way to save some settings and read them again, using a table structure in text format. https://www.rcgroups.com/forums/showthread.php?3129894-Easy-way-to-save-and-load-a-table-in-OpenTX-lua-scripts

How often do you want to change settings? let the user change the stored table. Not that difficult, but needs a pc, yes. (or phone with usb to sdcard file access)

azaz44, 2019-08-29:
> How often do you want to change settings? let the user change the stored table. Not that difficult, but needs a pc, yes. (or phone with usb to sdcard file access)

It depends on the application. In some of them - very often.

From my point of view, using a PC is totally unacceptable. Having a separate standalone script with UI  - this works, but has major disadvantages, as I outlined above.

Why not just allow for more parameters? Not sure about other radios, but on Horus X10 it seems it is 6 just because more would need scrolling the screen.

kilrah, 2019-08-29:
The limit is becasue we have to reserve number of options*number of possible scripts in model memory, which blows up quickly.

lshems, 2019-08-30:
Create a GUI in LUA to adapt the table you store. Not that difficult, and if properly done, can be generic, using the same structure as for inputs, and adapt automatically to the number of parameters.

closed 2020-05-26 by stale[bot]