opentx/opentx #6421: Feature Request:  Add auto power-off option for radios with power buttons

abertheham, 2019-05-13:
It would be fantastic if there was an option to automatically power-down radios like the X-lite and QX7, which have power buttons instead of physical switches.  

Back when I used a radio with a switch, it was habit to flip it off after landing and unplugging, but because of how long you have to hold down the power button to shut-down the radio, I now frequently leave mine on between packs or while working on the bench (as do several others I've talked to). 

There is an inactivity alarm, which is great if you're still near the radio, but I've left mine on and walked away only to come back to completely drained batteries more than once now.  I'd rather the radio just shut off if it hasn't been touched for a set amount of time - e.g. warn me after 10 minutes and shutdown after 12.  It's the one thing that drives me crazy on my x-lite, and addition of a feature like this would leave me with nothing to complain about.  

Thanks all your hard work!

Eldenroot, 2019-10-12:
Good idea, at least add it as an option. +1

salem83, 2020-01-12:
+1 

@abertheham 
Since 2.3, there's an option to set the power on/off delay (button press time) in the main radio settings. Additionally, you can disable the the splash-screen which saves also boot time. You may still leave the power on delay on 2s to prevent accidental tun on.

kilrah, 2020-01-12:
That's completely unrelated to the suggestion.

robertlacroix, 2020-07-03:
+1

It'd be great if the time to auto power down would be configurable and not shut down when a model is connected.

h2oreactor, 2020-07-30:
+1 on auto power off. 
Left my tx on a table and walked away.  I came back hearing the radio nagging "Inactivity Alarm" and realized it was doing that for 8hours non-stop. It will probably turn itself off at some point but not the way I would want it.

israelshirk, 2020-08-26:
I'd love to have this as well...  May scrap together a PR for it now that it's in my head and I almost killed a lipo today until Mike Chin said something in the lounge that reminded me to check...

howels, 2021-03-19:
+1 for this, just found my radio battery dead cos of this situation!

robertlacroix, 2021-09-22:
I'd say this is still relevant

bigtwin, 2021-09-22:
Just got noticed about this issue ... 
I probably would **not** use this feature and I currently dont miss an _auto-power-off_ at all. 
Therefore I highly appreciate if an _auto-power-off_ can be completely disabled, once it makes it into the OTX code.

pfeerick, 2021-09-22:
lol... Isn't that the very definition of it being an option? You can choose whether to use it or not, and what it should be set to? It is all about the implementation also - basic safety check can be made such as:
- no auto power off if telemetry link active
- no auto power off globally enabled, but inhibited in a specific model
- no auto power off if not enabled!
- configurable nag time warning you it will auto power off

robertlacroix, 2021-09-22:
> Just got noticed about this issue ... 
> 
> I probably would **not** use this feature and I currently dont miss an _auto-power-off_ at all. 
> 
> Therefore I highly appreciate if an _auto-power-off_ can be completely disabled, once it makes it into the OTX code.
> 

With the safety features @pfeerick lined out, would you still not use the feature? If not, why?

fwiw on multiple occasions I accidentally left my radio on after flying and the battery was totally dead the next day.

kleinerf, 2022-01-27:
Würde es auch begrüßen wenn so eine Funktion implementiert werden würde.