opentx/opentx #6432: Companion 2.3: "Error occurred while trying to play sound...(Err:[5])"

MikeNH, 2019-05-19:
**Describe the bug**
"Error occurred while trying to play sound...(Err:[5])" 
This error occurs when attempting to sample a wav file in the Editing Model page, Special Functions tab.

**To Reproduce**
1. With OpenTX configured for X-Lite Pro (or any Radio Type) create a new model using all defaults.
2 . Open the model document, Special Functions tab.
3. Choose Switch = FM0
4. Action = Play Track
5. Parameters = acro
6. Click play arrow. Error occurs

**Expected behavior**
The chosen wav file should play. Instead, there is the error above.

**General information**
-  OpenTX Companion 2.3 on Windows 10
- Same behavior despite chosen Radio Type or chosen wav file

raphaelcoeffic, 2019-08-18:
works here on OSX, must something Windows specific...

MikeNH, 2019-08-18:
Now, on 2,3,0N64 I agree, this is now working nicely. I opened this bug and am taking the liberty to close it.

closed 2019-08-18 by MikeNH