opentx/opentx #6433: OpenTX 2.2.3 Telemetry does not work with D8 protocol

BergshamraFraggel, 2019-05-20:
After updating from 2.1.9 to 2.2.3, telemetry does not work with D8 protocols.
You also get a warning when you bind recipients in D8 that the same receiver ID "  " is already used by other reciver.
Radio Taranis X9D non-EU
The receiver I tested is FrSky D4R-II.

3djc, 2019-05-20:
Have you rediscovered your sensors ?

BergshamraFraggel, 2019-05-20:
Yes, no sensors are found.
No RSSI or battery voltage for the receiver is displayed on the display.
Also did "Delete sensors" and discover again, found nothing.
Made bindings against a 4 in 1 module and the telemetry works.
With 4 in 1 module you can enter Receive ID also for D8 protocol.

3djc, 2019-05-20:
It does work fine, so there is probably something specific to your setup. Have you tried on a new model and make sure only internal module is selected ?

BergshamraFraggel, 2019-05-21:
It worked with new model memory and the old model worked after I copied it to a new memory location, very strange.
I guess something went wrong when converting EEPROM during the upgrade.

closed 2019-05-21 by BergshamraFraggel