opentx/opentx #6436: Companion 2.3: Splash screen blurred/corrupt

MikeNH, 2019-05-20:
[image]

**Describe the bug**
Companion 2.3 Startup/Splash screen displays incorrectly (See attached). This is a Windows 10 system. Companion 2.2 displays correctly.

**To Reproduce**
Starting Companion generates corrupted image.

**Screenshots**
Screenshot attached

**General information**
 - OpenTX version 2.3.0N9 May 20, 2019
 - Windows 10

bsongis, 2019-05-20:
We need an artist :)

mhotar, 2019-05-23:
[image]
Here we go. Now I have to set the git to be able to contribute to opentx repo again :)

3djc, 2019-05-23:
😍

mhotar, 2019-05-23:
[image]

3djc, 2019-05-23:
Commited !

closed 2019-05-23 by 3djc