opentx/opentx #6437: CSRF Vario Support

robthomson, 2019-05-20:
Can we please shedule in to work with TBS to get vario support running.

Essentially they will aim to send a new value  vSpd via crossfire module.

This is linked to this inav request:

https://github.com/iNavFlight/inav/issues/4710

3djc, 2019-05-20:
Sure, let me know how I can help

robthomson, 2019-05-20:
Brill.. thank you

closed 2019-06-04 by 3djc

teckel12, 2019-07-22:
@shellixyz @3djc I just tried this and I can only get positive values for vertical speed.  When I descend, it just shows `0` for `VSpd`

robthomson, 2019-07-22:
> @shellixyz @3djc I just tried this and I can only get positive values for vertical speed. When I descend, it just shows `0` for `VSpd`

I have been using this extensively from the latest version of inav with no issues. 

What xfire/inav versions you using?

kilrah, 2019-07-22:
Make sure you didn't check "only positive" in the sensor settings?

teckel12, 2019-07-22:
@robthomson I'm running 3.0 of crossfire and INAV 2.2.0.  I'll try it again and verify my results.  I can also upgrade to release 2.2.1, although I don't think there were any changes in INAV to Crossfire telemetry since INAV 2.1.0

@kilrah I just searched for new sensors and looked at the results, I don't believe "only positive" is checked by default, but I'll check to verify and let you know.

teckel12, 2019-07-23:
@kilrah Flew today and only positive values even though "only positive" wasn't checked. I have telemetry logs of you like. Not very interesting, but even a free fall is 0 f/s.

3djc, 2019-07-23:
As far as I can tell, our values are treated as int32, so no reason I can think off for those to be only positive, and since @robthomson is having no issue, that would seem to point either toward configuration and/or versions. We still do not have the Crossfire equipement to try to duplicate, that doesn't help

robthomson, 2019-07-23:
I have last night upgraded crossfire to the new V3 beta firmware.   This seems to work OK to on my setup.

3djc, 2019-07-23:
And just to fully confirm, you DO have vspeed negative values (sorry to ask this silly question, just want to make sure there is no chance for a misunderstanding) (and THANKS for your help)

teckel12, 2019-07-23:
@3djc I upgraded to INAV 2.2.1 and VSpd now shows negative values. Very odd as I didn't think there were any Crossfire changes in INAV this year, and for it to work but only for positive values was odd.

In any case, it works with INAV 2.2.1 so this issue can remain closed.

3djc, 2019-07-23:
Great, thanks for keeping us informed 👍

PR #6476 (2019-06-04): Add vario sensor to CRSF telemetry
Closes #6437 

Reference iNav issue: https://github.com/iNavFlight/inav/issues/4710