opentx/opentx #6440: Wrong PIN code shown for Horus' BT module

nmaggioni, 2019-05-21:
**Describe the bug**

The PIN code for associating with the BLE module on my X10S seems to be `000000` (six zeroes) instead of the [hardcoded](https://github.com/opentx/opentx/blob/c08417415bd8b5535b64c1b3e290edd10c99f161/radio/src/gui/480x272/radio_hardware.cpp#L210) `0000` (four zeroes) value in the hardware tab of the radio's settings.

**To Reproduce**
Steps to reproduce the behavior:
1. Set the Bluetooth module on `Telemetry` from the radio's hardware settings page.
2. Use an Android device to connect to a BLE service. [This currently under development app](https://github.com/CrazyDude1994/android-taranis-smartport-telemetry/issues/4#issuecomment-494515769) is what I'm testing at the moment.
3. Ignore the pairing request that appears on the phone since the BLE service doesn't require authentication.
4. Successfully communicate with the service.
5. When the authentication times out on the phone (a handful of seconds) the device disconnects.
6. Repeat step 2 but do not ignore the request, and pair with code `0000` as shown on the radio.
7. Pairing won't succeed.
8. Retry with code `000000` and the service will now stay connected.

**Expected behavior**
The PIN code shown on the radio should work when trying to pair a device with the radio's internal BT module.

**General information**
 - OpenTX 2.2.3 ([referenced code is still actual](https://github.com/opentx/opentx/blob/c08417415bd8b5535b64c1b3e290edd10c99f161/radio/src/gui/480x272/radio_hardware.cpp#L210))
 - FrSky Horus X10S

3djc, 2019-05-22:
Password seems indeed to be 000000, not 0000 will change the display

closed 2019-05-22 by 3djc

putimir, 2019-08-19:
Hello, I'm using my X10S with a recent 2.2.4, but I cannot pair my smartphone (Poco F1, Pie) neither with  0000 nor 000000 (Couldn't pair with X10S because of an incorrect PIN or passkey)

Any pointers?

nmaggioni, 2019-08-19:
@putimir How are you connecting? The BT module in the X10S is actually BLE: you cannot directly pair to it, you can just consume its services. Check the app I linked above for an example.

putimir, 2019-08-19:
Yes, that would be it, I can "pair" and connect with BLE scanner, sorry about that :) 

Thanks; learning everyday ;)

commit 115d92f8 (2019-05-22): Fix wrong BT password
Thix fixes #6440