opentx/opentx #6452: OpenTX 2.2.3 logical switch a<x not working

JoeDante, 2019-05-26:
I am using a Taranis QX7 with OpenTX 2.2.3 and a IrangeX multi protocol module with 2 Flysky quadcopters. 
Telemetry works fine, my QX7 discovers the sensors for the 1S Battery and displays the correct value as A1. I made a Special Function on the momentary switch with SH PlayVal A1, and when I flip the switch, the Taranis reads the value of the drone's battery. 

Now, I wanted to program a low battery alarm (for the drone's Lipo) but it won't work. 
I set a Logical Switch L1 a<x A1 3,2V and a special function for playing an alarm sound when the Lipo is under 3.2V, but the switch never gets avtivated. No matter how high or low I put the value, let's say 4.0V while the real value is displayed as 3.2V, the switch is not getting activatet. 
Is this an error? Am I missing something?

3djc, 2019-05-26:
Clearly, this logical switch function must be the most used of all, so it is extremely unlikely to be not working, just tried it on a X7, works fine. You must be doing something strange or wrong

kilrah, 2019-05-26:
It's very likely related to how the sensor's "Precision" is handled.
Try setting  a<x A1 0.4V and see if it activates when the battery is <4V?

JoeDante, 2019-05-26:
4V, -4V, 0,4V, doesn't matter. Switch is never activated....

kilrah, 2019-05-26:
Try 40 as well

JoeDante, 2019-05-26:
I tried +-40 and 400. 
But I just found out, it works with the Taranis internal battery. So I guess its the multi protocol module?

kilrah, 2019-05-26:
If you get the right value displayed it's unlikely, it's probably the handling of Flysky-specific telemetry in OpenTX. But most of us don't have FlySky equipment, so it's difficult for us to replicate/fix/test.

glowplugpropnut, 2019-09-24:
I have run into this exact same issue with a TinyWhoop Quad fitted with a FlySky receiver. I have also encountered it with a Hubsan Quad (obviously a different protocol). Both work perfectly with a multi-module on a radio using DeviationTX but not on my Taranis.