opentx/opentx #6453: X-VERT won't arm

fulltimer123, 2019-05-26:
I updated my IrangeX 4-in-1 module and now my AC will not arm.  No other changes made.

It worked perfectly before.

Taranis Plus running latest release of OpenTX.

Can't find anything on the DIY module github page

kilrah, 2019-05-26:
Nothing to do with us.

The module at some point changed scaling for DSM, could be that depending on what version you had before.

closed 2019-05-26 by kilrah

OpenUAS, 2020-08-10:
Maybe this will still help someone stumbling over this issue:
* https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/227
* https://www.youtube.com/watch?v=SDCpQijk9Is
* https://www.multi-module.org/using-the-module/protocol-details/dsm
* https://www.deviationtx.com/forum/model-configs/7234-horizon-hobby-x-vert-vtol#63430

Note that to  bind the TH_KILL has to be in killed position value at bind... else no bind possible