opentx/opentx #6459: ACCESS receiver with channel_range from 17_to_32 cannot select RX_OPT PWM output correctly

Ferdinando1968, 2019-05-28:
**Describe the bug**
I don't know if this is OpenTX related only, maybe module firmwares are responsable.
However, following is the test campus:

RADIO: XLITE-PRO with OpenTX 2.3.0N11 (and internal radio module with firmware ISRM_P_ACCESS_1.1.1.frk)
RECEIVER: G-RX8 (with ACCESS firmware FW-G-RX8_ACCESS_v1.1.1)

with receiver already registered to RADIO and binded to new model (binded with ACCESS using internal radio module),
in model menu (1), modify ISRM channel_range from 1_to_16 to 17_to_32,
than select receiver-->options to change the channel for one of the specific PWM outputs (8),
so, instead of correct channels values selectable (like 17,,32),
incorrect channel values are displayed (seems randoms values: inputs, telemetry, etc)

**Expected behavior**
For each of 8 PWM outputs, you should can choose channel from 17 to 32

**General information**
RADIO: XLITE-PRO with OpenTX 2.3.0N11 (and internal radio module with firmware ISRM_P_ACCESS_1.1.1.frk)
RECEIVER: G-RX8 (with ACCESS firmware FW-G-RX8_ACCESS_v1.1.1)

Ferdinando1968, 2019-05-30:
further tests I have made shows that the (wrong) behavior happens as soon as
TX channel range is moved from default of 1..16 to 1+x..16+x.
I still not tested 1+x..16.

bsongis, 2019-05-31:
Reproduced

commit 59e2347a (2019-05-31): Fixes #6459
closed 2019-05-31 by bsongis