opentx/opentx #6470: "Can't extract RADIO/radio.bin"

MikeNH, 2019-05-30:
**Describe the bug**
Using Companion setup for XLite Pro unable to open a otx file created with the previous nightly

**To Reproduce**
1 - Install N12 (in my case)
2 - Create a new model for multirotor using only defaults. I named the model "N12"
3 - Install latest nightly. I installed N19.
3 - With N19 nightly, attempt to open N12 model.
Error occurs "Can't extract RADIO/radio.bin"

**Expected behavior**
I should be able to open a model created in a previous Companion version. Lost all configuration.

**General information**
 - OpenTX version N19
 - Hardware Frsky XLite Pro
 - OpenTX Companion N19 on Windows 10

kilrah, 2019-05-30:
Nightlies are automated development snapshots, anything can be broken in them. Please try saving and reopening a file with latest version, only if that doesn't work will there be a valid issue.

bsongis, 2019-05-31:
Would you attach your otx file?

MikeNH, 2019-06-01:
Sure thing... please find it attached.

kilrah, 2019-06-01:
Github doesn't support mail attachments.

MikeNH, 2019-06-02:
[N12.txt](https://github.com/opentx/opentx/files/3245330/N12.txt)
[N19.txt](https://github.com/opentx/opentx/files/3245331/N19.txt)

Change the txt extension to otx.  I had to rename so it would upload.

The N12 wascreated by the N12 nightly. The N19 was created by the current nightly.

notlebowski, 2019-08-21:
I'm running 2.3 nightlies on an X-Lite classic and have an OTX of almost each week.
7 weeks ago I migrated my 2.2.4 nightly to 2.3 nightlies and had models corrupted by the radio.
I was able to correct this by going back to 2.2.4 and migrate the setup again with a 2.3 version that was fixed.
Later I found some other changes in the setup that were probably caused by Companion 2.3.
I corrected those changes manually on the radio.

I now would like to read one of my 2.2.4 setups with 2.3 Companion and do a compare with my current setup to verify my setup.
Companion however is unable to read a 2.2.4 setup and will give a "Can't extract RADIO/radio.bin"

Is this something that is going to be fixed before the 2.3 release?
If 2.3 is able to read 2.2.x setups it can be used to find those changes easier because testers can check their setups.

bsongis, 2019-08-21:
Yes, it will be fixed

closed 2020-02-18 by kilrah