opentx/opentx #6472: 2.3 missing popupInput

3djc, 2019-06-02:
Got put aside and needs to be readded

raphaelcoeffic, 2019-08-18:
@3djc any news here? What was the reason why it was removed / deactivated?

3djc, 2019-08-18:
@bsongis removed it because popups have been completely redesigned. It needs to be rewritten to support the new popups

raphaelcoeffic, 2019-08-26:
@bsongis @3djc I had a look at where it is used, and it seems that internally, it is only used with the curve presets. In LUA, I cannot imagine someone using it, as it is really not nice for entering a parameter. So probably everyone made it differently.

Bottom line: what about just obsoleting that function?

3djc, 2019-08-26:
I opened it because I got informed that some fairly well known script was using it and failed with 2.3. And of course, I cannot remember which one :(

3djc, 2019-08-26:
Exemple is JFrickmann's DLG timing script

bsongis, 2019-08-26:
please would you share the script so that we can think of a good solution?

raphaelcoeffic, 2019-08-26:
@jfrickmann Jesper? Could you please jump in so we can find a good solution for your script?

raphaelcoeffic, 2019-08-26:
@jfrickmann thx!

lshems, 2019-08-28:
You can write a lua version of the popup input. Wasn't it used in the wizard? must be some code to be reused there than.

jfrickmann, 2019-08-28:
I plan to do it the same way I have done the J programs - no problem!

————————
Jesper Frickmann

closed 2020-06-06 by stale[bot]

sande005, 2020-06-10:
I'd suggest this be reopened.  I was using it on a telemetry script that tracked flight time and battery usage.  Used the popupInput to allow entry for however many mah were used for recharge during a day of flying.  X9d running 2.3.7