opentx/opentx #6475: opentx-9x-frsky-en-2.0.19 does not read AIL and THR switches

teodoryantcheff, 2019-06-03:
Modded 9x with frsky telemetry mod - confirmed hardware working properly and all.

Both `opentx-9x-en-2.0.19.hex` and `opentx-9x-frsky-en-2.0.19.hex` behave identically -- `THR` and `AIL` switches read **HIGH** all the time.

**To Reproduce**
Steps to reproduce the behavior:
1. Install firmware version `opentx-9x-en-2.0.19.hex` or `opentx-9x-frsky-en-2.0.19.hex`
2. `THR` and `AIL` read on all the time. Both switches **do not work**.

Since I know it's a properly modded transmitter went back to `er9x.hex` and `er9x-frsky.hex`.
As expected `er9x.hex` cannot read `AIL` and `THR` and `er9x-frsky.hex` **works** as expected.

`opentx-9x-frsky-en-2.1.9.hex` **also works properly!** - `THR` and `AIL` switches work and are read as expected. 

**2.0.19** is the offending version. Could not get older firmwares from the 2.0.x series to see how far this goes.

**General information**
 - OpenTX version 2.0.20
 - Hardware: **Modded flysky 9x** 
 - OpenTX Companion: **2.0**

kilrah, 2019-06-03:
2.0 support was discontinued long ago, use 2.1.9.

teodoryantcheff, 2019-06-05:
@kilrah Thanks.

closed 2019-06-05 by teodoryantcheff

Vonaris, 2021-03-01:
Hi. I have the same problem. I checked my wiring twice with multimeter and seems good. HTR and ALI i have still HIGH. I can't enter also to telemetry screen. Is someone can share me checked right working firmware for rutnigy 9x with frsky mod ?