opentx/opentx #6481: Xlite Pro and MPM Elite issue

Talha909, 2019-06-06:
The issue tracker here is for devlopment of OpenTX. This is not a
support site, only a bug tracking system. Please use support groups
like [rcgroups](https://rcgroups.com/) or
[openrcforum](https://www.openrcforums.com/forum/). See also our
[manual](https://opentx.gitbooks.io/manual-for-opentx-2-2/).

For quick questions you can also join our general chat.

Issues that fall into the category of asking for help, will be
generally closed as invalid without an answer.

Talha909, 2019-06-06:
I have updated my xLite Pro with the latest update of power management and to Open tx Nightly build. I have one problem with my MPM external module. When I turn on the radio . The red and green light turn on my MPM and some times its stay green. When it shows the red and green then i switch between the models for the change of light. when switching between the model its comes again green. Whats the reason behind for red and green light. I have already update the power management unit
--

fabioeh, 2019-06-08:
Having same Issue.

MPM Vantech module Fails to go online. Status light turns Red. Will not Bind.

Setting External RF to off and reselecting MULTI brings the MPM module online and the Status light turns green. Turning radio off and on will not bring MPM online. 

Using DSM x22 protocol.

MPM ver 1.2.1.51,  Opentx N21, Firmware 190605

3djc, 2019-06-08:
We do not believe this is linked to OpenTX, but to FkSky PMU chip, which we know virtually nothing about (it is not 'driven' or managed by OpenTX).

kilrah, 2019-06-09:
Do you have internal module on or off on that model?

Talha909, 2019-06-09:
No. I only on and off the external module . If i turn on and off extertanl model then the green light comes.

kilrah, 2019-06-09:
You did not answer the question...

Talha909, 2019-06-09:
I have off the internal module when i Used the external module.

fabioeh, 2019-06-09:
I updated to PMU chip with the latest firmware from the FRSKY web site but it did not solve the problem.

3djc, 2019-06-10:
Our prototypes do not have the PMU chip, and I cannot replicate this issue, it works perfectly with N22 (or earlier too). When internal module is ON, you just loose telemetry, which is to be expected since two module are sharing the sport line. I'm leaving this open in case someone brings a new idea or sympthom to the table, but for now, I would assume it is not an OpenTX issue, but an FrSky one

Talha909, 2019-06-10:
Its works works if you got the green light on the Module.  Green light did not comes first.  I have to switch model on my Xlite Pro. I have add 4 models and bind with the d16.  I had switch off the internal module and switch on the external module on all four . If i did not see the green then I switch between models . I do it 3 or 4 times. After that light blink and green light switch on. And then its ok.

3djc, 2019-06-10:
We understand, but since it doesn’t do that at all on our radio without PMU, it must be linked to PMU, therefore there is nothing we can do. Please contact your dealer or FrSky

closed 2019-06-10 by 3djc

fabioeh, 2019-06-10:
Hi 3djc 

When will you be able to test OpenTX on a production model with the PMU

This what uses a purchasing not prototypes.

3djc, 2019-06-10:
That’s not likely to happen, but the only difference between the two is PMU. So we know clearly it comes from that, and OpenTX has no control or access to it. Only FrSky can do something about it.

fabioeh, 2019-06-10:
Thanks for the response. Hard to believe the X-Lite-pro will be working any time soon since the OpenTx team has no access to test with the MPU. And yore team have no contact or the ability to refer problems to FRSky and the process is just close the problem unsolved and not reported to the vendor.

3djc, 2019-06-11:
FrSky is aware of the issue, but it's all in their hand. I can keep the issue open, but since there is nothing we can do about it, there is little  point in that ?

fabioeh, 2019-06-11:
Thanks. Will wait for FRSKY to fix the problem.

Talha909, 2019-06-12:
solution is here

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/237