opentx/opentx #6485: companion simulator: program collabs when modellname has maximum signs

Blackforrestpilot, 2019-06-08:
Hi,
when the modellname has a lenght of 15, start of simulator will crash companion.
**To reproduce:**
Take a modellname with a lenght of 14 signs (e.g. High_End_Pro_M) --> simulator will work
Take a modellname with a lenght of 15 signs (e.g. High_End_Pro_Mo) --> simulator will crash

**General information**
 - OpenTX version 2.2.3
 - Hardware FrSky X12
 - OpenTX Companion Version 2.2.3
 - Win10 + Win7

kilrah, 2019-12-06:
Can't reproduce on 2.3.

Blackforrestpilot, 2019-12-06:
I've tested with companion 2.3.2, Nov 16 2019 under Win7
The issue is allmost present.
[Test_Simulator.zip](https://github.com/opentx/opentx/files/3932707/Test_Simulator.zip)
The OTX in the zip containes the same model two times, with different names: 12 and 15 letters.
At my system companion crashes by starting the simulator with model with 15 letters, the other model run without trouble.
Not a big problem, when you know about.

Best regards
Christoph

kilrah, 2019-12-07:
Reproduced with provided eeprom

bsongis, 2019-12-07:
Fixed by 74db396c364f942352ea09f3de4dc60cdd3aedb0

closed 2019-12-07 by bsongis