opentx/opentx #6493: mirror SPORT telemetry on SPORT TX pins on new radio without JR module

Ferdinando1968, 2019-06-12:
I don't know if technically possible, so I ask.

As you know, new FrSky radios has not traditional JR module
(nor the serial pins in battery compartment as TARANIS has),
and new external modules does not have pins to get SPORT telemetry.

However, new radios have the sport pins, usually used to flash firmware on receivers or so.
For what I know, these pins carries no signal, unless specifically triggered by human action "flash sport".

Is it possible to have an option, in model menu for example, to get sport pins output (permanently) telemetry,
so as to make it possible to have telemetry as I have on old FrSky JR XJT/R9M modules ?

Thanks.

kilrah, 2019-06-12:
Both the module slot and the sport connector carry the sport signal anytime one is present just like on older models.

Ferdinando1968, 2019-06-12:
Kilrah, as I have written, I don't know if technically possible.
But I have thinked that if SPORT signal/information il still present inside radio,
why not using it, outputting it on SPORT pin,
instead using that pin only to flash receivers ?
That is what I meant with "permanently" (I meants not exclusively).

Maybe the sport signal is always present on relative pin,
also when I do not "flash sport" (I have not verified, sorry),
but the power pin (on sport connector) seems goes
down when do not "flash sport".
Thanks for your time.

Ferdinando1968, 2019-12-09:
Hi.
I'm still interested in this feature.
It should permit to use telemetry module like iMsb with xlites tx series.
No one is interested ?

zosko, 2020-02-27:
Im interested,
i have Taranis X9D Plus 2019, right pins in battery bay is S.Port VCC GND. 
From iNav im getting telemetry from Smart Port to taranis, now i want to make S.Port to MavLink to transfer it to Antenna Tracker, but without success, seems that pins are not working anymore.

brainbubblersbest, 2020-11-01:
related to #7723 ?

brainbubblersbest, 2020-12-15:
#8073