opentx/opentx #6498: xlite-pro: touching "max bauds" in radio setting (menu 6/7), it will power on external module

Ferdinando1968, 2019-06-13:
xlite-pro: touching "max bauds" in radio setting (menu 6/7), it will power on external TX module,
even if the current model has the external tx configured as OFF.
I don't know which signal the external module receive in this situation.
I don't know if power is stable on external tx in this way.
maybe dangerous ?

**To Reproduce**
1. create a model with ext TX configured as OFF (internal ISRM does not care)'
2. go in radio menu menu 6/7, and hit enter on "max bauds"
3. the external TX will power-on.
   to shut down it, you must edit model, set ext tx as on and after set it to  off

**Expected behavior**
change the max baud-rate to external tx 

**General information**
 - OpenTX version 2.3.0.N23
 - Hardware xlite-pro with internal tx latest fw (ISRM_PRO_190604.frk)
   and PMU patched (PMU.frsk)

bsongis, 2019-06-13:
wow, I take it

Ferdinando1968, 2019-06-13:
strange behavior !
maybe "PMU" related ?

bsongis, 2019-06-14:
Reproduced. I think this one is in OpenTX.

bsongis, 2019-06-14:
Thanks for reporting this issue, it was present on all radios. As it's rather major I will port the fix to 2.2 as well

commit efdeb6bb (2019-06-14): Fixes #6498
commit d1289b0f (2019-06-14): Fixes #6498
closed 2019-06-14 by bsongis