opentx/opentx #6499: R9M full size on X-Lite Pro

playlet, 2019-06-13:
Hi, guys!

I just tried R9M full size module on X-Lite Pro, on latest N24 nightly.
It binds and works ok but there are some issues, described below.

Latest FLEX firmware on 868 has the following power levels: 10mw, 100, 500, adaptive.
Nightly for X-Lite Pro has 25 8ch, 25 16ch, 200, and 500. Do these power levels correspond to the above mentioned for flex: 10, 100, 500, 1W (adaptive)?

Next, on nightly, 200 and 500 options don't have the option to turn the telemetry on while binding (only 1-8 off and 9-16 off). So if I bind on 200 or 500mw, there is no telemetry.
Latest flex firmware enables telemetry on all power levels so it should work.
Workaround: bind on 25mw with telemetry on and then switch to higher power levels. It works.

Third problem (might not be a problem but only a fluke).
After binding, when I was trying different power levels, somehow 200mw was buzzing louder on my X9D Plus speaker nearby than 500mw.
Now, after each new try, it seems ok and 500mw is louder than 200mw. Strange, can't reproduce it anymore.

Just wanted to share stuff I noticed.

Thanks for your support.

Kind regards,

Ivan

3djc, 2019-06-14:
Not sure what you tried !

[image]

playlet, 2019-06-14:
Hi!

I downloaded N024 latest nightly companion form here: https://downloads.open-tx.org/2.3/nightlies/companion/windows/

Have it setup like this:
[image]
[image]

Downloaded nightly N024:
[image]

Installed:
[image]
[image]

Setup new model, and this is what I get in the menu:
[image]
[image]
[image]
[image]

Did I miss anything?

Kind regards,

Ivan

3djc, 2019-06-14:
What was the full name of the downloaded firmware ?

playlet, 2019-06-14:
opentx-xlites-flexr9m-lua-luac-noheli-en-2.3.0N24

There is no other option, only FCC or EU with R9M.

3djc, 2019-06-14:
There seems to be something really strange happening on the compilation server, will investigate tomorow

playlet, 2019-06-14:
Thanks for the support, I am at your disposal for any further tests from my side.

3djc, 2019-06-14:
Yes, you can try N25 tomorow, I have pushed a fix

3djc, 2019-06-15:
Tested this morning, now ok :)

closed 2019-06-15 by 3djc

playlet, 2019-06-15:
Confirmed, it's ok. Thanks!